momenso/node-dht-sensor #11: Can't run - node-dht-sensor.node: undefined symbol: init

taggalucci, 2015-01-17:
I'm having trouble with running this package. When I run the example code for a single DTH11 sensor, I get the following:
```javascript
module.js:485                                                                                                                                                     
  process.dlopen(filename, module.exports);                                                                                                                       
          ^                                                                                                                                                       
Error: /usr/share/adafruit/webide/node_modules/node-dht-sensor/build/Release/node-dht-sensor.node: undefined symbol: init                                         
    at Object.Module._extensions..node (module.js:485:11)                                                                                                         
    at Module.load (module.js:356:32)                                                                                                                             
    at Function.Module._load (module.js:312:12)                                                                                                                   
    at Module.require (module.js:362:17)                                                                                                                          
    at require (module.js:378:17)                                                                                                                                 
    at Object.<anonymous> (/usr/share/adafruit/webide/repositories/my-pi-projects/DHT11.js:1:79)                                                                  
    at Module._compile (module.js:449:26)                                                                                                                         
    at Object.Module._extensions..js (module.js:467:10)                                                                                                           
    at Module.load (module.js:356:32)                                                                                                                             
    at Function.Module._load (module.js:312:12)
```
I haven't been able to figure this out. Can anyone help please?

momenso, 2015-01-17:
What node.js version are you using?

taggalucci, 2015-01-18:
Thanks for your quick response!

After your question, I've discovered that I have 2 versions:

0.8.17 at /usr/local/bin/node
0.6.19 at /usr/bin/node

I currently use Adafruit's WebIDE (which is supposed to have come with 0.8) and I can see that I manually installed 0.8.17 when I was attempting to get Cloud9 working (as per the instructions [here](http://www.bernaschina.com/en/blog/raspberrypi/cloud9-raspberry-pi/)). So perhaps this is causing issues?

I didn't bother backing out of the attempt to get Cloud9 working - not sure whether this would be a worthwhile exercise or not. It hasn't caused me issues with anything else (so far).

momenso, 2015-01-18:
These are old versions of node.js. I recommend you use a newer release...
The recommended way to get the latest version of node.js is the following.
``` shell
$ wget http://node-arm.herokuapp.com/node_latest_armhf.deb 
$ sudo dpkg -i node_latest_armhf.deb
```
Using node.js v0.8 I was able to reproduce the same problem but on a newer release you should have no problems.

taggalucci, 2015-01-19:
Thanks! this has now resolved the issue.

closed 2015-01-20 by momenso