momenso/node-dht-sensor #12: Install fails with undeclared identifier sched_setscheduler

joscha, 2015-01-20:
I installed the BCM2385 library, but on `npm install` I get the following error
```console
> npm install node-dht-sensor --save
> node-dht-sensor@0.0.6 install /Users/joscha/Development/rpi-dht22-logger/node_modules/node-dht-sensor
> node-gyp rebuild

  CXX(target) Release/obj.target/node-dht-sensor/node-dht-sensor.o
../node-dht-sensor.cpp:180:5: error: use of undeclared identifier 'sched_setscheduler'
    sched_setscheduler(0, SCHED_FIFO, &schedp);
    ^
1 error generated.
make: *** [Release/obj.target/node-dht-sensor/node-dht-sensor.o] Error 1
```
any idea what I might be doing wrong?

momenso, 2015-01-20:
It does look like the BCM2835 library header is not right.

If you go to the folder where you extracted the file __bcm2835-1.xx.tar.gz__ and run the following command does it report test passed? Do not forget to use **sudo**.

```bash
~/bcm2835-1.38 $ sudo make check
...
============================================================================
Testsuite summary for bcm2835 1.38
============================================================================
# TOTAL: 1
# PASS:  1
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
```

if it fails, I recommend you follow all the instructions from [http://www.airspayce.com/mikem/bcm2835/](http://www.airspayce.com/mikem/bcm2835/) in order to make sure the library is installed correctly.

joscha, 2015-01-21:
Thanks! It seems to work on the raspberry pi directly - probably just a local problem, closing this for now.

closed 2015-01-21 by joscha