momenso/node-dht-sensor #13: can't install

sclausen, 2015-02-14:
I have libbcm in 1.39
nodejs in v0.12.0 and it can't be installed

    pi@raspberrypi ~/dht22 $ npm install --save node-dht-sensor
    npm WARN package.json dht22@0.0.1 No description
    npm WARN package.json dht22@0.0.1 No repository field.
    npm WARN package.json dht22@0.0.1 No README data
    |
    > node-dht-sensor@0.0.6 install /home/pi/dht22/node_modules/node-dht-sensor
    > node-gyp rebuild

    child_process: customFds option is deprecated, use stdio instead.
    make: Entering directory '/home/pi/dht22/node_modules/node-dht-sensor/build'
      CXX(target) Release/obj.target/node-dht-sensor/node-dht-sensor.o
    ../node-dht-sensor.cpp:205:26: error: ‘Arguments’ does not name a type
    ../node-dht-sensor.cpp:205:37: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h: In function ‘v8::Handle<v8::Value> Read(const int&)’:
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:816:13: error: ‘v8::HandleScope::HandleScope()’ is protected
    ../node-dht-sensor.cpp:206:17: error: within this context
    ../node-dht-sensor.cpp:216:41: error: no matching function for call to ‘v8::Object::New()’
    ../node-dht-sensor.cpp:216:41: note: candidate is:
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:2388:24: note: static v8::Local<v8::Object> v8::Object::New(v8::Isolate*)
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:2388:24: note:   candidate expects 1 argument, 0 provided
    ../node-dht-sensor.cpp:217:18: error: ‘NewSymbol’ is not a member of ‘v8::String’
    ../node-dht-sensor.cpp:217:69: error: no matching function for call to ‘v8::Number::New(float&)’
    ../node-dht-sensor.cpp:217:69: note: candidate is:
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:1999:24: note: static v8::Local<v8::Number> v8::Number::New(v8::Isolate*, double)
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:1999:24: note:   candidate expects 2 arguments, 1 provided
    ../node-dht-sensor.cpp:218:18: error: ‘NewSymbol’ is not a member of ‘v8::String’
    ../node-dht-sensor.cpp:218:75: error: no matching function for call to ‘v8::Number::New(float&)’
    ../node-dht-sensor.cpp:218:75: note: candidate is:
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:1999:24: note: static v8::Local<v8::Number> v8::Number::New(v8::Isolate*, double)
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:1999:24: note:   candidate expects 2 arguments, 1 provided
    ../node-dht-sensor.cpp:219:18: error: ‘NewSymbol’ is not a member of ‘v8::String’
    ../node-dht-sensor.cpp:219:72: error: no matching function for call to ‘v8::Boolean::New(bool)’
    ../node-dht-sensor.cpp:219:72: note: candidate is:
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:6194:17: note: static v8::Handle<v8::Boolean> v8::Boolean::New(v8::Isolate*, bool)
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:6194:17: note:   candidate expects 2 arguments, 1 provided
    ../node-dht-sensor.cpp:220:18: error: ‘NewSymbol’ is not a member of ‘v8::String’
    ../node-dht-sensor.cpp:220:68: error: no matching function for call to ‘v8::Number::New(int)’
    ../node-dht-sensor.cpp:220:68: note: candidate is:
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:1999:24: note: static v8::Local<v8::Number> v8::Number::New(v8::Isolate*, double)
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:1999:24: note:   candidate expects 2 arguments, 1 provided
    ../node-dht-sensor.cpp:221:18: error: ‘class v8::HandleScope’ has no member named ‘Close’
    ../node-dht-sensor.cpp: At global scope:
    ../node-dht-sensor.cpp:224:30: error: ‘Arguments’ does not name a type
    ../node-dht-sensor.cpp:224:41: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h: In function ‘v8::Handle<v8::Value> ReadSpec(const int&)’:
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:816:13: error: ‘v8::HandleScope::HandleScope()’ is protected
    ../node-dht-sensor.cpp:225:17: error: within this context
    ../node-dht-sensor.cpp:227:14: error: request for member ‘Length’ in ‘args’, which is of non-class type ‘const int’
    ../node-dht-sensor.cpp:228:44: error: ‘New’ is not a member of ‘v8::String’
    ../node-dht-sensor.cpp:228:85: error: ‘ThrowException’ was not declared in this scope
    ../node-dht-sensor.cpp:229:21: error: ‘class v8::HandleScope’ has no member named ‘Close’
    ../node-dht-sensor.cpp:229:37: error: too few arguments to function ‘v8::Handle<v8::Primitive> v8::Undefined(v8::Isolate*)’
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:305:28: note: declared here
    ../node-dht-sensor.cpp:232:28: error: invalid types ‘const int[int]’ for array subscript
    ../node-dht-sensor.cpp:234:45: error: ‘New’ is not a member of ‘v8::String’
    ../node-dht-sensor.cpp:234:94: error: ‘ThrowException’ was not declared in this scope
    ../node-dht-sensor.cpp:235:22: error: ‘class v8::HandleScope’ has no member named ‘Close’
    ../node-dht-sensor.cpp:235:38: error: too few arguments to function ‘v8::Handle<v8::Primitive> v8::Undefined(v8::Isolate*)’
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:305:28: note: declared here
    ../node-dht-sensor.cpp:241:49: error: ‘New’ is not a member of ‘v8::String’
    ../node-dht-sensor.cpp:241:86: error: ‘ThrowException’ was not declared in this scope
    ../node-dht-sensor.cpp:242:26: error: ‘class v8::HandleScope’ has no member named ‘Close’
    ../node-dht-sensor.cpp:242:42: error: too few arguments to function ‘v8::Handle<v8::Primitive> v8::Undefined(v8::Isolate*)’
    /home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:305:28: note: declared here
    ../node-dht-sensor.cpp:246:22: error: invalid types ‘const int[int]’ for array subscript
    ../node-dht-sensor.cpp:255:41: error: no matching function for call to ‘v8::Object::New()’
    ../node-dht-sensor.cpp:255:41: note: candidate is:
[... 85 lines cut ...]
    gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
    gyp ERR! cwd /home/pi/dht22/node_modules/node-dht-sensor
    gyp ERR! node -v v0.12.0
    gyp ERR! node-gyp -v v1.0.2
    gyp ERR! not ok 
    npm ERR! Linux 3.18.5+
    npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "--save" "node-dht-sensor"
    npm ERR! node v0.12.0
    npm ERR! npm  v2.5.1
    npm ERR! code ELIFECYCLE

    npm ERR! node-dht-sensor@0.0.6 install: `node-gyp rebuild`
    npm ERR! Exit status 1
    npm ERR! 
    npm ERR! Failed at the node-dht-sensor@0.0.6 install script 'node-gyp rebuild'.
    npm ERR! This is most likely a problem with the node-dht-sensor package,
    npm ERR! not with npm itself.
    npm ERR! Tell the author that this fails on your system:
    npm ERR!     node-gyp rebuild
    npm ERR! You can get their info via:
    npm ERR!     npm owner ls node-dht-sensor
    npm ERR! There is likely additional logging output above.

    npm ERR! Please include the following file with any support request:
    npm ERR!     /home/pi/dht22/npm-debug.log

momenso, 2015-02-14:
This was due to the breaking changes introduced by **node.js v0.12**. I have just published an update (node-dht-sensor@*0.0.7*) to fix this issue. Once you update to the latest *0.0.7* version you should have no more problems.

In case you have problems to get the latest version, then try cleaning your npm cache (i.e. ```npm clean cache```).

sclausen, 2015-02-15:
yay thanks! :+1:

closed 2015-02-15 by sclausen