momenso/node-dht-sensor #14: Cannot install

HSkilgannon, 2015-02-21:
I have installed libcm 1.39
Am i Installing it wrong? Please Help

```npm WARN engine node-dht-sensor@0.0.7: wanted: {"node":">=0.12 <0.13"} (current:{"node":"0.10.36","npm":"1.4.28"}```
```
 >node-dht-sensor@0.0.7 install /home/pi/Desktop/node-red-0.10.1/node_modules/node-dht-sensor
 >node-gyp rebuild

make: Entering directory '/home/pi/Desktop/node-red-0.10.1/node_modules/node-dht-sensor/build'
  CXX(target) Release/obj.target/node-dht-sensor/node-dht-sensor.o
../node-dht-sensor.cpp:205:17: error: ‘FunctionCallbackInfo’ does not name a type
../node-dht-sensor.cpp:205:17: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../node-dht-sensor.cpp:205:37: error: expected ‘,’ or ‘...’ before ‘<’ token
../node-dht-sensor.cpp: In function ‘void Read(int)’:
../node-dht-sensor.cpp:207:30: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../node-dht-sensor.cpp:207:30: note: candidates are:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:473:3: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘const v8::HandleScope&’
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:448:3: note: v8::HandleScope::HandleScope()
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:448:3: note:   candidate expects 0 arguments, 1 provided
../node-dht-sensor.cpp:217:48: error: no matching function for call to ‘v8::Object::New(v8::Isolate*&)’
../node-dht-sensor.cpp:217:48: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1728:33: note: static v8::Local<v8::Object> v8::Object::New()
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1728:33: note:   candidate expects 0 arguments, 1 provided
../node-dht-sensor.cpp:218:18: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:218:89: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, float&)’
../node-dht-sensor.cpp:218:89: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note: static v8::Local<v8::Number> v8::Number::New(double)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note:   candidate expects 1 argument, 2 provided
../node-dht-sensor.cpp:219:18: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:219:95: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, float&)’
../node-dht-sensor.cpp:219:95: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note: static v8::Local<v8::Number> v8::Number::New(double)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note:   candidate expects 1 argument, 2 provided
../node-dht-sensor.cpp:220:18: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:220:92: error: no matching function for call to ‘v8::Boolean::New(v8::Isolate*&, bool)’
../node-dht-sensor.cpp:220:92: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:4364:17: note: static v8::Handle<v8::Boolean> v8::Boolean::New(bool)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:4364:17: note:   candidate expects 1 argument, 2 provided
../node-dht-sensor.cpp:221:18: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:221:88: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, int)’
../node-dht-sensor.cpp:221:88: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note: static v8::Local<v8::Number> v8::Number::New(double)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note:   candidate expects 1 argument, 2 provided
../node-dht-sensor.cpp:223:5: error: ‘args’ was not declared in this scope
../node-dht-sensor.cpp: At global scope:
../node-dht-sensor.cpp:226:21: error: ‘FunctionCallbackInfo’ does not name a type
../node-dht-sensor.cpp:226:21: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../node-dht-sensor.cpp:226:41: error: expected ‘,’ or ‘...’ before ‘<’ token
../node-dht-sensor.cpp: In function ‘void ReadSpec(int)’:
../node-dht-sensor.cpp:228:30: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../node-dht-sensor.cpp:228:30: note: candidates are:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:473:3: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘const v8::HandleScope&’
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:448:3: note: v8::HandleScope::HandleScope()
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:448:3: note:   candidate expects 0 arguments, 1 provided
../node-dht-sensor.cpp:230:9: error: ‘args’ was not declared in this scope
../node-dht-sensor.cpp:231:15: error: ‘class v8::Isolate’ has no member named ‘ThrowException’
../node-dht-sensor.cpp:232:10: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:236:22: error: ‘args’ was not declared in this scope
../node-dht-sensor.cpp:238:18: error: ‘class v8::Isolate’ has no member named ‘ThrowException’
../node-dht-sensor.cpp:239:10: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:246:19: error: ‘class v8::Isolate’ has no member named ‘ThrowException’
../node-dht-sensor.cpp:247:11: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:261:48: error: no matching function for call to ‘v8::Object::New(v8::Isolate*&)’
../node-dht-sensor.cpp:261:48: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1728:33: note: static v8::Local<v8::Object> v8::Object::New()
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1728:33: note:   candidate expects 0 arguments, 1 provided
../node-dht-sensor.cpp:262:18: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:262:89: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, float&)’
[... 60 lines cut ...]
gyp ERR! System Linux 3.18.7+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/Desktop/node-red-0.10.1/node_modules/node-dht-sensor
gyp ERR! node -v v0.10.36
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok 

npm ERR! node-dht-sensor@0.0.7 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-dht-sensor@0.0.7 install script.
npm ERR! This is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.18.7+
npm ERR! command "/usr/local/bin/node" "/usr/local/bin/npm" "install" "node-dht-sensor"
npm ERR! cwd /home/pi/Desktop/node-red-0.10.1
npm ERR! node -v v0.10.36
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0)```

momenso, 2015-02-21:
If you are using node.js < 0.12 then you can use *node-dht-sensor@0.0.6* which is the previous version. The latest version **0.0.7** was updated for node.js 0.12.0 that is not backwards compatible due to breaking changes that were introduced. 

So you can either update to node.js v0.12 or use the previous versions but you'll have to use node-dht-sensor@0.0.6.

HSkilgannon, 2015-02-21:
Thanks,
I was able to install it on Raspberry b.

closed 2015-02-21 by HSkilgannon