momenso/node-dht-sensor #15: Compatibility with RasPi 2?

stozk, 2015-02-21:
Hi,

did anyone try this on the RasPi 2 already?

I've setup a fresh system, but it will just freeze/hang when running the sample app.
Only connectin via putty will make it respond again, ctrl+c won't do anything, ctrl-alt-del will restart.
When I put the SD card in my RasPi B+ model it will work normally.

BR
Daniel

momenso, 2015-02-21:
I haven't had a chance to get my hands on a RPi 2 yet. But it is great that you have!
Because RPi 2 uses the more advanced BCM2836 you better try to use the newer library v1.39!
Please try installing the newer version from the link below and let us know if it works! 

[BCM Library v1.39](http://www.airspayce.com/mikem/bcm2835/bcm2835-1.39.tar.gz)

stozk, 2015-02-21:
Ah nice thanks, I didn't see that there is a newer version.
It seems to work again now with v1.39

BR
Daniel

closed 2015-02-24 by stozk