momenso/node-dht-sensor #16: Memory problem

HSkilgannon, 2015-02-23:
I have encountered a problem when running this node

After a while I get the following:

bcm2835_init: gpio mmap failed: Cannot allocate memory

Has this happened to you? I have the newest bcm library 1.39.
Any sugestions?
Thank you

momenso, 2015-02-23:
Can you share your code? 
I'd like to replicate the problem to try to understand what is going on...

HSkilgannon, 2015-02-24:
I'm using raspberry pi B (the one with 26pins and 2 USB only)
Raspian
Node JS 0.12.0
Node red 0.10.1
bcm library 1.39

It crashes after 5 minutes running every time. 

Node Red currently does not support node js 0.12.0. So i don't know if the problem has something to do with that or with node-dht-sensor or even bcm.

The code inside the nodes is pretty simple:

```
var sensorLib = context.global.dht;
var load;
var temperature;
sensorLib.initialize(11, 4);
var readout = sensorLib.read();
 temperature=parseInt(readout.temperature.toFixed(2));

var humidity;
var readout = sensorLib.read();
 humidity=readout.humidity.toFixed(2);
 context.global.data.d.humidity=humidity;
```

I have the following flow on node red:
```
[{"type":"tab","id":"8a13c7b0.7932a","label":"Sheet 1"},{"id":"f811f5d1.f6ae8","type":"subflow","name":"Subflow 1","in":[],"out":[]},{"id":"59c26074.1ced5","type":"function","name":"Toggle LED on input","func":"\n// select wpi pin 0 = pin 11 on header (for v2)\nvar pin = 0;\n\n// initialise the wpi to use the global context\nvar wpi = context.global.wpi;\n\n// use the default WiringPi pin number scheme...\nwpi.setup();\n\n// initialise the state of the pin if not already set\n// anything in context.  persists from one call to the function to the next\ncontext.state = context.state || wpi.LOW;\n\n// set the mode to output (just in case)\nwpi.pinMode(pin, wpi.modes.OUTPUT);\n\n// toggle the stored state of the pin\n(context.state == wpi.LOW) ? context.state = wpi.HIGH : context.state = wpi.LOW;\n\n// output the state to the pin\nwpi.digitalWrite(pin, context.state);\n\n// we don't \"need\" to return anything here but may help for debug\nreturn msg;","outputs":1,"x":560,"y":353,"z":"f811f5d1.f6ae8","wires":[["261d06f0.766772"]]},{"id":"2c12ba3d.1f68a6","type":"inject","name":"tick","topic":"","payload":"","payloadType":"date","repeat":"1","once":false,"x":340,"y":333,"z":"f811f5d1.f6ae8","wires":[["59c26074.1ced5"]]},{"id":"261d06f0.766772","type":"debug","name":"","active":true,"x":780,"y":373,"z":"f811f5d1.f6ae8","wires":[]},{"id":"80315c78.3ae85","type":"debug","name":"debug json string","active":true,"console":"false","complete":"false","x":656,"y":354.91668701171875,"z":"8a13c7b0.7932a","wires":[]},{"id":"61c81c53.0cef8c","type":"inject","name":"Trigger","topic":"","payload":"\"Hello World\"","payloadType":"none","repeat":"1","crontab":"","once":false,"x":161,"y":326.91668701171875,"z":"8a13c7b0.7932a","wires":[["f32a3ec5.88335"]]},{"id":"f32a3ec5.88335","type":"function","name":"DHT 11 temp","func":"if (!context.global.hasOwnProperty(\"data\")) {\n\tcontext.global.data = {};\n\tcontext.global.data.d = {\n\t\t\"myName\" : \"DHT\"\n\t};\n}\nvar sensorLib = context.global.dht;\nvar load;\nvar temperature;\nsensorLib.initialize(11, 4);\nvar readout = sensorLib.read();\n temperature=parseInt(readout.temperature.toFixed(2));\n\n\ncontext.global.data.d.temperature = temperature;\n\nmsg.payload = context.global.data;\nreturn msg;\n","outputs":1,"x":389,"y":287.91668701171875,"z":"8a13c7b0.7932a","wires":[["27fa9275.80ced6"]]},{"id":"27fa9275.80ced6","type":"function","name":"DHT 11 temp Hum","func":"\nvar sensorLib = context.global.dht;\n\nvar humidity;\nsensorLib.initialize(11, 4);\nvar readout = sensorLib.read();\n humidity=readout.humidity.toFixed(2);\n context.global.data.d.humidity=humidity;\n\n\nmsg.payload = context.global.data;\nreturn msg;\n","outputs":1,"x":393,"y":360.91668701171875,"z":"8a13c7b0.7932a","wires":[["80315c78.3ae85"]]}]
```

momenso, 2015-02-25:
You're right. I can reproduce the problem. It really crashes after 5 minutes.

The good news is that it is not hard to fix the problem. What is happening is that the function **initialize** is called every time the sensor is read and this is supposed to be done only once.

Because the same code is called a bunch of times, we can just use another function of the _node-dht-sensor_ library that does not require initialization, namely **readSpec**

A simplified code would be the following:
```javascript
if (!context.global.hasOwnProperty("data")) {
	context.global.data = {};
}

var sensorLib = context.global.dht;
var readout = sensorLib.readSpec(11, 4);
context.global.data.temperature = readout.temperature.toFixed(0);
context.global.data.humidity = readout.humidity.toFixed(0);

msg.payload = context.global.data;
return msg;
```

Please have a try with the flow below and see if it works for you. 

```javascript
[{"id":"80315c78.3ae85","type":"debug","name":"debug json string","active":true,"console":"false","complete":"false","x":541,"y":326.91668701171875,"z":"8a13c7b0.7932a","wires":[]},{"id":"61c81c53.0cef8c","type":"inject","name":"Trigger","topic":"","payload":"\"Hello World\"","payloadType":"none","repeat":"1","crontab":"","once":false,"x":161,"y":326.91668701171875,"z":"8a13c7b0.7932a","wires":[["f32a3ec5.88335"]]},{"id":"f32a3ec5.88335","type":"function","name":"DHT 11","func":"if (!context.global.hasOwnProperty(\"data\")) {\n\tcontext.global.data = {};\n}\n\nvar sensorLib = context.global.dht;\nvar readout = sensorLib.readSpec(11, 4);\ncontext.global.data.temperature = readout.temperature.toFixed(0);\ncontext.global.data.humidity = readout.humidity.toFixed(0);\n\nmsg.payload = context.global.data;\nreturn msg;\n","outputs":1,"x":342,"y":326.91668701171875,"z":"8a13c7b0.7932a","wires":[["80315c78.3ae85"]]}]
```

I have this code running for a while and I had no problems.

HSkilgannon, 2015-02-26:
I have tested it for a while and it works perfectly!

Thank you!

closed 2015-02-26 by HSkilgannon

sbigulo, 2016-10-30:
Hi Nomenso, I have the same memory leak problem, i apply your workaround but receive the error TypeError: Cannot call method 'readSpec' of undefined this is my enviroment Node.js  version: v0.10.29 Linux 4.4.21+ arm LE ode-RED version: v0.15.2.  thank in advance. 
Danilo.

momenso, 2016-10-30:
That's referring to an old API. You can call the ```read``` function instead of ```readSpec```.
See if #52 helps.

emilutwww, 2019-09-19:
I have same problem, I can't rid of this issue with bcm2835, 

the problem is: bcm2835_init: gpio mmap failed: Cannot allocate memory
Segmentation fault

I want to connect at QIoT Qnap.
I have a rpi4,I've installed the last version of bcm, 1.6, I tried another version like 1.59 and 1.46 and nothing change. Please, help me guys! Thanks!

momenso, 2019-09-19:
I will have a look at this issue once I get my hands on a RPi 4. 
Work will be done on #101.