momenso/node-dht-sensor #17: Issue when i try "sudo npm install node-dht-sensor" (after install BCM2835 library)

rloreto, 2015-04-14:
When i try to intall this error:

sudo npm install node-dht-sensor

> node-dht-sensor@0.0.7 install /home/pi/node_modules/node-dht-sensor
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.12.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/node-dht-sensor/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: node-v0.12.0.tar.gz local checksum b73cf11c8e1df999f2fea8ce59dd1ecd25df699c2fd352610cc3ba05c915857b not match remote 9700e23af4e9b3643af48cef5f2ad20a1331ff531a12154eef2bfb0bb1682e32
gyp ERR! stack     at deref (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:299:20)
gyp ERR! stack     at IncomingMessage.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:340:13)
gyp ERR! stack     at IncomingMessage.emit (events.js:129:20)
gyp ERR! stack     at _stream_readable.js:908:16
gyp ERR! stack     at process._tickCallback (node.js:355:11)
gyp ERR! System Linux 3.18.5+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/node-dht-sensor
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok
npm ERR! Linux 3.18.5+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "node-dht-sensor"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE

npm ERR! node-dht-sensor@0.0.7 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-dht-sensor@0.0.7 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/node-v0.12.0/npm-debug.log

Can someone help me?

momenso, 2015-04-14:
No need to use **sudo** for installing. Using ```npm install node-dht-sensor``` should suffice.
Administrative privileges should be needed only when running your app.

cristiancs, 2015-05-01:
I got the same error without using sudo

gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok
npm ERR! Linux 3.12.35+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "node-dht-sensor"
npm ERR! node v0.12.1
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE

npm ERR! node-dht-sensor@0.0.7 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-dht-sensor@0.0.7 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.

nicor88, 2015-05-01:
Did you install BCM 2835 library (http://www.airspayce.com/mikem/bcm2835/)?

cristiancs, 2015-05-01:
thanks, problem solved.

closed 2015-05-12 by momenso

pakair, 2018-08-06:
I get the same error as @cristiancs posted -- but I have Library bcm2835 installed. (the 1.56 version). Is there a way to clean up npm and start again?