momenso/node-dht-sensor #18: Incorrect Temp and Humidity values

hawkerboy7, 2015-05-01:
Hi thanks for this module!

However I do get strange temp and humidity values:
```
info  DH11        →  Temperature: 588.80C, humidity: 870.40% 
```

When I breath very close to the sensor the humidity increases to over 1400% and temp around 716.80C.
Do you know what went wrong?

Some images on google showed a resistor had to be placed in between pin 1 and 2 (4k7). I did this and also removed it but the values seemed to stay the same so it didn't appear to have any effect.

Do you know what's wrong?

Thanks in advance!

momenso, 2015-05-01:
I've seen this happen before when I switched between the *DHT22* to the *DHT11* sensor without changing the sensor type setting in my code.

This happen because although the library can handle both *DHT11* and *DHT22* sensor you need to specify which one are you using. They are compatible but use different communication protocols which need different handling in order to get the correct readout value.

So, first thing you should check your code if either the ```initialize``` or ```readSpec``` function call is specifying the correct sensor type (i.e. ```11``` for the *DHT11* and ```22``` for the *DHT22*). Please note that the first example in the README file uses an *DHT22* (aka AM2302).

Also regarding _automake_ and the ```m4``` directory, you should not need to worry about these if you're using a clean install of raspian and bcm2834 v1.44. Is that what you have?

hawkerboy7, 2015-05-02:
Thank you! I am getting good readings now. I used 22 and not 11 as you said.

Yes I used a clean install of raspian but not bcm2834 v1.44.
On the [website (bcm2835)](http://www.airspayce.com/mikem/bcm2835/) it says on line (or <p>) 5 that you should use this link: http: www.airspayce.com/mikem/bcm2835/bcm2835-1.42.tar.gz.

But now it works great! Thanks! =D

closed 2015-05-02 by hawkerboy7