momenso/node-dht-sensor #21: check-bcm2835 failure

hurricnz, 2015-05-20:
Can't install the sensor module

here is the log file
https://www.dropbox.com/s/ny2lcv8zn0y2h12/npm-debug.log?dl=0

nicor88, 2015-05-20:
It looks that you didn't install bcm2835. Try to install it using this link:
http://www.airspayce.com/mikem/bcm2835/

hurricnz, 2015-05-20:
I have bmc2835-1.8 installed so that is not the problem

nicor88, 2015-05-20:
'Failed at the node-dht-sensor@0.0.8 install script './check-bcm2835'.'
Btw it looks that you have a problem with this lib!

hurricnz, 2015-05-20:
I was following this post https://www.openhomeautomation.net/monitor-your-home-raspberry-pi-b/

Which points to a link for the library which is "dead" 
I looked for another library online and found the 1.8 version. Maybe there is something wrong with the library. Could you supply me wit the right one?

Reinstalling dident work

nicor88, 2015-05-20:
I used 1.44 version! Download it from here: 
http://www.airspayce.com/mikem/bcm2835/bcm2835-1.44.tar.gz
I hope that with this lib works!

hurricnz, 2015-05-20:
Ok thanks for the help!
Ill try it first thing in the morning, im heading off for now

hurricnz, 2015-05-21:
I guess I couldent wait. it worked. just a faulty library :D don't use 1.8

closed 2015-05-22 by momenso