momenso/node-dht-sensor #23: Failed to install module.

sanket-mindstix, 2015-08-03:
Getting an error during the installation of this module. I have installed  BCM2835 library though module is giving an error during installation. I have executed ./check-bcm2835 script, it gives me a following output,

Library bcm2835 found.
gyp info it worked if it ends with ok
gyp info using node-gyp@2.0.2
gyp info using node@0.12.6 | linux | arm
gyp http GET https://nodejs.org/dist/v0.12.6/node-v0.12.6.tar.gz
gyp http 200 https://nodejs.org/dist/v0.12.6/node-v0.12.6.tar.gz
gyp http GET https://nodejs.org/dist/v0.12.6/SHASUMS256.txt
gyp http 200 https://nodejs.org/dist/v0.12.6/SHASUMS256.txt
gyp info spawn python2
gyp info spawn args [ '/usr/lib/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/pi/.npm/node-dht-sensor/0.0.8/package/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/lib/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/pi/.node-gyp/0.12.6/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/pi/.node-gyp/0.12.6',
gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/node-gyp',
gyp info spawn args   '-Dmodule_root_dir=/home/pi/.npm/node-dht-sensor/0.0.8/package',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory `/home/pi/.npm/node-dht-sensor/0.0.8/package/build'
  CXX(target) Release/obj.target/node-dht-sensor/node-dht-sensor.o
../node-dht-sensor.cpp: In function ‘long int readDHT(int, int, float&, float&)’:
../node-dht-sensor.cpp:75:22: error: ‘bcm2835_delay’ was not declared in this scope
../node-dht-sensor.cpp:92:36: error: ‘bcm2835_delayMicroseconds’ was not declared in this scope
make: *** [Release/obj.target/node-dht-sensor/node-dht-sensor.o] Error 1
make: Leaving directory `/home/pi/.npm/node-dht-sensor/0.0.8/package/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.10.25+
gyp ERR! command "node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/.npm/node-dht-sensor/0.0.8/package
gyp ERR! node -v v0.12.6
gyp ERR! node-gyp -v v2.0.2
gyp ERR! not ok

momenso, 2015-08-03:
What version of the library BCM2835 have you installed?

sanket-mindstix, 2015-08-03:
1.5 version

momenso, 2015-08-03:
It seems that newer versions of this lib have changed.

Can you try version 1.44?

http://www.airspayce.com/mikem/bcm2835/bcm2835-1.44.tar.gz

sanket-mindstix, 2015-08-03:
Thank module has been installed.

sanket-mindstix, 2015-08-11:
Hi,

Can you please check what errors this module is giving, I have attached
error log file with a mail, following are the node and npm versions I am
using,

node- v0.10.26
npm- 1.4.3

Thanks.

sanket-mindstix, 2015-08-11:
I am getting following error logs during installation of this module,

npm http GET https://registry.npmjs.org/node-dht-sensor
npm http 200 https://registry.npmjs.org/node-dht-sensor
npm http GET https://registry.npmjs.org/node-dht-sensor/-/node-dht-sensor-0.0.8.tgz
npm http 200 https://registry.npmjs.org/node-dht-sensor/-/node-dht-sensor-0.0.8.tgz
npm WARN engine node-dht-sensor@0.0.8: wanted: {"node":">=0.12 <0.13"} (current: {"node":"v0.10.26","npm":"1.4.3"})

> node-dht-sensor@0.0.8 install /home/pi/node_modules/node-dht-sensor
> ./check-bcm2835

Library bcm2835 found.
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.10.26"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/node-dht-sensor/.node-gyp"
gyp http GET http://nodejs.org/dist/v0.10.26/node-v0.10.26.tar.gz
gyp http 200 http://nodejs.org/dist/v0.10.26/node-v0.10.26.tar.gz
make: Entering directory '/home/pi/node_modules/node-dht-sensor/build'
  CXX(target) Release/obj.target/node-dht-sensor/node-dht-sensor.o
../node-dht-sensor.cpp:205:17: error: ‘FunctionCallbackInfo’ does not name a type
../node-dht-sensor.cpp:205:17: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../node-dht-sensor.cpp:205:37: error: expected ‘,’ or ‘...’ before ‘<’ token
../node-dht-sensor.cpp: In function ‘void Read(int)’:
../node-dht-sensor.cpp:207:30: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../node-dht-sensor.cpp:207:30: note: candidates are:
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:473:3: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘const v8::HandleScope&’
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:448:3: note: v8::HandleScope::HandleScope()
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:448:3: note:   candidate expects 0 arguments, 1 provided
../node-dht-sensor.cpp:217:48: error: no matching function for call to ‘v8::Object::New(v8::Isolate*&)’
../node-dht-sensor.cpp:217:48: note: candidate is:
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:1724:33: note: static v8::Local<v8::Object> v8::Object::New()
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:1724:33: note:   candidate expects 0 arguments, 1 provided
../node-dht-sensor.cpp:218:18: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:218:89: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, float&)’
../node-dht-sensor.cpp:218:89: note: candidate is:
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:1377:33: note: static v8::Local<v8::Number> v8::Number::New(double)
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:1377:33: note:   candidate expects 1 argument, 2 provided
../node-dht-sensor.cpp:219:18: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:219:95: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, float&)’
../node-dht-sensor.cpp:219:95: note: candidate is:
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:1377:33: note: static v8::Local<v8::Number> v8::Number::New(double)
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:1377:33: note:   candidate expects 1 argument, 2 provided
../node-dht-sensor.cpp:220:18: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:220:92: error: no matching function for call to ‘v8::Boolean::New(v8::Isolate*&, bool)’
../node-dht-sensor.cpp:220:92: note: candidate is:
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:4349:17: note: static v8::Handle<v8::Boolean> v8::Boolean::New(bool)
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:4349:17: note:   candidate expects 1 argument, 2 provided
../node-dht-sensor.cpp:221:18: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../node-dht-sensor.cpp:221:88: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, int)’
../node-dht-sensor.cpp:221:88: note: candidate is:
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:1377:33: note: static v8::Local<v8::Number> v8::Number::New(double)
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:1377:33: note:   candidate expects 1 argument, 2 provided
../node-dht-sensor.cpp:223:5: error: ‘args’ was not declared in this scope
../node-dht-sensor.cpp: At global scope:
../node-dht-sensor.cpp:226:21: error: ‘FunctionCallbackInfo’ does not name a type
../node-dht-sensor.cpp:226:21: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../node-dht-sensor.cpp:226:41: error: expected ‘,’ or ‘...’ before ‘<’ token
../node-dht-sensor.cpp: In function ‘void ReadSpec(int)’:
../node-dht-sensor.cpp:228:30: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../node-dht-sensor.cpp:228:30: note: candidates are:
/home/pi/node_modules/node-dht-sensor/.node-gyp/0.10.26/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
[... cut ...]
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.12.28+
npm ERR! command "/opt/node/bin/node" "/opt/node/bin/npm" "install" "node-dht-sensor"
npm ERR! cwd /home/pi
npm ERR! node -v v0.10.26
npm ERR! npm -v 1.4.3
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/npm-debug.log
npm ERR! not ok code 0

Node.js and npm versions I am using on raspberry pi device is
0.10.26 and 1.4.3 respectively,
bcm2835  Library version is 1.44

momenso, 2015-08-11:
You should not use node-dht-sensor@0.0.8 with older versions of node (i.e. before v12) that´s why you are getting these errors.
To resolve this, you should either upgrade node to v12+ or use an older version of node-dht-sensor (e.g. v0.0.6).

sanket-mindstix, 2015-08-11:
Great thanks, the issue has been resolved.

closed 2015-12-02 by momenso