momenso/node-dht-sensor #26: Can't install

fiws, 2015-08-28:
```
/
> node-dht-sensor@0.0.8 install /home/fiws/test/node_modules/node-dht-sensor
> ./check-bcm2835

Library bcm2835 found.
make: Entering directory '/home/fiws/test/node_modules/node-dht-sensor/build'
  CXX(target) Release/obj.target/node-dht-sensor/node-dht-sensor.o
  SOLINK_MODULE(target) Release/obj.target/node-dht-sensor.node
/usr/bin/ld: /usr/local/lib/libbcm2835.a(bcm2835.o): relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libbcm2835.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
node-dht-sensor.target.mk:110: recipe for target 'Release/obj.target/node-dht-sensor.node' failed
make: *** [Release/obj.target/node-dht-sensor.node] Error 1
make: Leaving directory '/home/fiws/test/node_modules/node-dht-sensor/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/fiws/npm/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 4.1.6-1-ARCH
gyp ERR! command "node" "/home/fiws/npm/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/fiws/test/node_modules/node-dht-sensor
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 
npm ERR! Linux 4.1.6-1-ARCH
npm ERR! argv "node" "/home/fiws/npm/bin/npm" "install" "node-dht-sensor"
npm ERR! node v0.12.7
npm ERR! npm  v2.13.0
npm ERR! code ELIFECYCLE

npm ERR! node-dht-sensor@0.0.8 install: `./check-bcm2835`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-dht-sensor@0.0.8 install script './check-bcm2835'.
npm ERR! This is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./check-bcm2835
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/fiws/test/npm-debug.log
```

momenso, 2015-08-28:
What version of lib bcm2835 are you using?

fiws, 2015-08-29:
I did add this to my issue:
> I've tested with bcm2835 version 1.45, 1.44 and 1.5. All fail, but 1.5 has a different error.

momenso, 2015-08-30:
I just tested with the following and found no issues.
- Raspberry Pi Model B running 2015-05-05-raspbian-wheezy.img, 
- node v0.12.6 / npm 2.11.2
- bcm2835 v1.45
- node-dht-sensor v0.0.8

Are you still having problems?

rtepowers, 2015-09-03:
From the instructions:
```bash
wget  http://www.airspayce.com/mikem/bcm2835/bcm2835-1.45.tar.gz
```
Then compile bcm
```bash
tar zxvf bcm2835-1.45.tar.gz
cd bcm2835-1.45
./configure
make
sudo make check
sudo make install
```
Then you should be good to go for:
```bash
npm install node-dht-sensor
```

fiws, 2015-09-03:
Thats how I tried it. 
Full output:
```
λ tau ~ → cd /tmp 

[omitting downloading and extracting]

λ tau /tmp → cd bcm2835-1.45
λ tau bcm2835-1.45 → ./configure 
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
/tmp/bcm2835-1.45/missing: Unknown `--is-lightweight' option
Try `/tmp/bcm2835-1.45/missing --help' for more information
configure: WARNING: 'missing' script is too old or missing
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking dependency style of gcc... gcc3
checking for clock_gettime in -lrt... yes
checking for doxygen... no
configure: WARNING: Doxygen not found - continuing without Doxygen support
checking for ranlib... ranlib
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking dependency style of gcc... (cached) gcc3
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating doc/Makefile
config.status: creating config.h
config.status: executing depfiles commands
λ tau bcm2835-1.45 → make
make  all-recursive
make[1]: Entering directory '/tmp/bcm2835-1.45'
Making all in src
make[2]: Entering directory '/tmp/bcm2835-1.45/src'
gcc -DHAVE_CONFIG_H -I. -I..     -g -O2 -MT bcm2835.o -MD -MP -MF .deps/bcm2835.Tpo -c -o bcm2835.o bcm2835.c
mv -f .deps/bcm2835.Tpo .deps/bcm2835.Po
rm -f libbcm2835.a
ar cru libbcm2835.a bcm2835.o 
ranlib libbcm2835.a
make[2]: Leaving directory '/tmp/bcm2835-1.45/src'
Making all in doc
make[2]: Entering directory '/tmp/bcm2835-1.45/doc'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/tmp/bcm2835-1.45/doc'
make[2]: Entering directory '/tmp/bcm2835-1.45'
make[2]: Leaving directory '/tmp/bcm2835-1.45'
make[1]: Leaving directory '/tmp/bcm2835-1.45'
λ tau bcm2835-1.45 → sudo make check
[sudo] password for fiws: 
Making check in src
make[1]: Entering directory '/tmp/bcm2835-1.45/src'
make  test
make[2]: Entering directory '/tmp/bcm2835-1.45/src'
gcc -DHAVE_CONFIG_H -I. -I..     -g -O2 -MT test.o -MD -MP -MF .deps/test.Tpo -c -o test.o test.c
[... 81 lines cut ...]
gyp ERR! cwd /tmp/node_modules/node-dht-sensor
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 
npm ERR! Linux 4.1.6-2-ARCH
npm ERR! argv "node" "/home/fiws/npm/bin/npm" "install" "node-dht-sensor"
npm ERR! node v0.12.7
npm ERR! npm  v2.13.0
npm ERR! code ELIFECYCLE

npm ERR! node-dht-sensor@0.0.8 install: `./check-bcm2835`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-dht-sensor@0.0.8 install script './check-bcm2835'.
npm ERR! This is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./check-bcm2835
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/npm-debug.log
```

brunnokick, 2015-11-29:
Hi, I am having the same problem, did you fix it ?

momenso, 2015-11-30:
This problem has to do with BCM2835 library that is not being compiled correctly to work on your linux distro. I suggest you recompile BCM library using the following procedure.

```shell
# start at home
$ cd ~

# download the bcm2835 library package
$ wget http://www.airspayce.com/mikem/bcm2835/bcm2835-1.46.tar.gz

# extract the package
$ tar zxvf bcm2835-1.46.tar.gz

# compile the package
cd bcm2835-1.46
./configure
make clear
make CFLAGS='-g -O2 -fPIC'
sudo make check
sudo make install
```

Also, note that BCM2835 requires device tree to be enabled in the kernel. So, try to see if you have ```/proc/device-tree/soc/ranges```, if not then that means you need to enable device tree support on the kernel. I am not familiar with Arch Linux, but you can find more information here: https://gist.github.com/bodokaiser/6854708

brunnokick, 2015-12-01:
Thanks for helping, but it didn't work. I'm using Raspbian Jessie.
This is what is happening when I do make install

````
Mking install in src
make[1]: Entering directory '/home/pi/bcm2835-1.46/src'
make[2]: Entering directory '/home/pi/bcm2835-1.46/src'
 /bin/mkdir -p '/usr/local/lib'
 /usr/bin/install -c -m 644  libbcm2835.a '/usr/local/lib'
 ( cd '/usr/local/lib' && ranlib libbcm2835.a )
 /bin/mkdir -p '/usr/local/include'
 /usr/bin/install -c -m 644 bcm2835.h '/usr/local/include'
make[2]: Leaving directory '/home/pi/bcm2835-1.46/src'
make[1]: Leaving directory '/home/pi/bcm2835-1.46/src'
Making install in doc
make[1]: Entering directory '/home/pi/bcm2835-1.46/doc'
make[2]: Entering directory '/home/pi/bcm2835-1.46/doc'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/bcm2835-1.46/doc'
make[1]: Leaving directory '/home/pi/bcm2835-1.46/doc'
make[1]: Entering directory '/home/pi/bcm2835-1.46'
make[2]: Entering directory '/home/pi/bcm2835-1.46'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/bcm2835-1.46'
make[1]: Leaving directory '/home/pi/bcm2835-1.46'
````

I tried deleting the 'libbcm2835.a' which is created in /user/local/lib, didn't work neither.

momenso, 2015-12-01:
I don't see any problems on your output of the ```make install``` - it looks all normal.
Did you see any errors when you run ```sudo make check```? If not, you should be good to go.
After completing BCM2835 library installation you should be able to install the node-dht-sensor via npm.

brunnokick, 2015-12-02:
Hello again, there s nothing wrong with ````make check````, this is my npm-debug.log. Tha last thing I remember doing was installing express via npm.

````
0 info it worked if it ends with ok
1 verbose cli [ 'node', '/usr/bin/npm', 'install', 'node-dht-sensor' ]
2 info using npm@3.5.0
3 info using node@v0.10.29
4 verbose node symlink /usr/bin/node
5 silly loadCurrentTree Starting
6 silly install loadCurrentTree
7 silly install readLocalPackageData
8 silly fetchPackageMetaData node-dht-sensor
9 silly fetchNamedPackageData node-dht-sensor
10 silly mapToRegistry name node-dht-sensor
11 silly mapToRegistry using default registry
12 silly mapToRegistry registry https://registry.npmjs.org/
13 silly mapToRegistry uri https://registry.npmjs.org/node-dht-sensor
14 verbose request uri https://registry.npmjs.org/node-dht-sensor
15 verbose request no auth needed
16 info attempt registry request try #1 at 20:36:04
17 verbose request id e14d7a155d19b3c8
18 verbose etag "EP0I8633FS5VBCEW9Q292W55U"
19 http request GET https://registry.npmjs.org/node-dht-sensor
20 http 304 https://registry.npmjs.org/node-dht-sensor
21 verbose headers { date: 'Wed, 02 Dec 2015 20:36:05 GMT',
21 verbose headers   via: '1.1 varnish',
21 verbose headers   'cache-control': 'max-age=60',
21 verbose headers   etag: '"EP0I8633FS5VBCEW9Q292W55U"',
21 verbose headers   age: '0',
21 verbose headers   connection: 'keep-alive',
21 verbose headers   'x-served-by': 'cache-atl6225-ATL',
21 verbose headers   'x-cache': 'MISS',
21 verbose headers   'x-cache-hits': '0',
21 verbose headers   'x-timer': 'S1449088565.645114,VS0,VE397',
21 verbose headers   vary: 'Accept' }
22 silly get cb [ 304,
22 silly get   { date: 'Wed, 02 Dec 2015 20:36:05 GMT',
22 silly get     via: '1.1 varnish',
22 silly get     'cache-control': 'max-age=60',
22 silly get     etag: '"EP0I8633FS5VBCEW9Q292W55U"',
22 silly get     age: '0',
22 silly get     connection: 'keep-alive',
22 silly get     'x-served-by': 'cache-atl6225-ATL',
22 silly get     'x-cache': 'MISS',
22 silly get     'x-cache-hits': '0',
22 silly get     'x-timer': 'S1449088565.645114,VS0,VE397',
22 silly get     vary: 'Accept' } ]
23 verbose etag https://registry.npmjs.org/node-dht-sensor from cache
24 verbose get saving node-dht-sensor to /root/.npm/registry.npmjs.org/node-dht-sensor/.cache.json
25 silly install normalizeTree
26 silly loadCurrentTree Finishing
27 silly loadIdealTree Starting
28 silly install loadIdealTree
29 silly cloneCurrentTree Starting
30 silly install cloneCurrentTreeToIdealTree
31 silly cloneCurrentTree Finishing
32 silly loadShrinkwrap Starting
33 silly install loadShrinkwrap
34 silly loadShrinkwrap Finishing
35 silly loadAllDepsIntoIdealTree Starting
36 silly install loadAllDepsIntoIdealTree
37 silly resolveWithNewModule node-dht-sensor@0.0.8 checking installable status
38 silly cache add args [ 'node-dht-sensor', null ]
39 verbose cache add spec node-dht-sensor
40 silly cache add parsed spec { raw: 'node-dht-sensor',
40 silly cache add   scope: null,
40 silly cache add   name: 'node-dht-sensor',
40 silly cache add   rawSpec: '',
40 silly cache add   spec: 'latest',
40 silly cache add   type: 'tag' }
[... 1877 lines cut ...]
302 verbose stack     at EventEmitter.emit (events.js:98:17)
302 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:24:14)
302 verbose stack     at ChildProcess.emit (events.js:98:17)
302 verbose stack     at maybeClose (child_process.js:755:16)
302 verbose stack     at Process.ChildProcess._handle.onexit (child_process.js:822:5)
303 verbose pkgid node-dht-sensor@0.0.8
304 verbose cwd /var/working/node
305 error Linux 4.1.13+
306 error argv "node" "/usr/bin/npm" "install" "node-dht-sensor"
307 error node v0.10.29
308 error npm  v3.5.0
309 error code ELIFECYCLE
310 error node-dht-sensor@0.0.8 install: `./check-bcm2835`
310 error Exit status 1
311 error Failed at the node-dht-sensor@0.0.8 install script './check-bcm2835'.
311 error Make sure you have the latest version of node.js and npm installed.
311 error If you do, this is most likely a problem with the node-dht-sensor package,
311 error not with npm itself.
311 error Tell the author that this fails on your system:
311 error     ./check-bc2835
311 error You can get their info via:
311 error     npm owner ls node-dht-sensor
311 error There is likely additional logging output above.
312 verbose exit [ 1, true ]
``

momenso, 2015-12-02:
Installation via NPM is failing at ***check-bcm2835***. In that case, after ensuring that BCM2835 is installed correctly, the only other thing to check is if *node-gyp* is installed. If not, you can install it via ```npm install node-gyp -g```.

denisbrondy, 2016-01-05:
hi,
Sorry but I can't make it installed... The BCM2835 install appears ok, the node-gyp too. What is the next step to get the node package installed as the npm install node-dht-sensor still fails ?
Tks for your help
denis

momenso, 2016-01-05:
If BCM library is installed correctly, then I'd suggest you to install the recommended node.js version, like so: 

```bash
$ wget http://node-arm.herokuapp.com/node_archive_armhf.deb
$ sudo dpkg -i node_archive_armhf.deb
```

fiws, 2016-01-06:
@momenso got it working with your instructions to compile bmc! :+1:

aziraphale, 2016-02-02:
Yeah so I've just spent ages trying to get this to install, only to finally have it install by using @momenso's [method](https://github.com/momenso/node-dht-sensor/issues/26#issuecomment-169133502) - i.e. explicitly using node v0.12.6, from July 2015, rather than either v4.2.x LTS or v5.x Stable as downloaded from nodejs.org, or using whatever ships with Raspbian (Wheezy or Jessie). This is a huge pain in the arse, especially as this seems to also require the use of npm v2.11.2, which is similarly-old and really quite rubbish compared to newer releases.

Is there any chance of getting this library updated to work with newer nodejs versions at some point? Sadly I'm not familiar enough with the lower-level elements of nodejs to be able to contribute usefully, but I'm happy to donate a bit, if that's an option?

Personally I would much rather add a dependency on WiringPi (#10) and then have things actually working and installing on a wider range of nodejs versions, than rely on the BCM library, which not only requires you to go and compile that completely separately from installing the library via npm, but may also be related to the failure to install on newer nodejs versions (I don't know if that's the case or not)...

momenso, 2016-02-02:
Yes, we should be able to upgrade to current version of node.js - the tricky bit will be to not break compatibility with older deployments.

As for wiringPi we already had a pull request for it but it still used BCM library underneath, so it didn't really simplified the dependency and also changed the pin numbering scheme which affects already deployed project settings.

Given we have two possibly breaking changes we might go for a experimental version with the goals of having a simplified installation and compatibility with current node.js.

How that sounds? I'd really appreciate your help with testing and general feedback.

roland-vachter, 2016-03-10:
Any news for supporting Node v4? I'm struggling with the same errors as mentioned in this thread ...

momenso, 2016-03-10:
Yes, the support is already available in the current version _node-dht-sensor@0.0.11_.

fiws, 2016-03-10:
I can confirm that this works with the latest node.
I compiled bmc manually like @momenso suggested:
```
# start at home
$ cd ~

# download the bcm2835 library package
$ wget http://www.airspayce.com/mikem/bcm2835/bcm2835-1.46.tar.gz

# extract the package
$ tar zxvf bcm2835-1.46.tar.gz

# compile the package
cd bcm2835-1.46
./configure
make clear
make CFLAGS='-g -O2 -fPIC'
sudo make check
sudo make install
```

So I'm gonna close this issue
[image]

closed 2016-03-10 by fiws

fiws, 2016-03-10:
actually, I got another problem with 0.0.11 – but i guess that's another problem, so I'm opening another issue.

techk9solutions, 2016-09-08:
I've read all the above and followed process described.  Here is what I get now.  Seems to be related to node-dht-sensor:

/usr/share/adafruit/webide/repositories/my-pi-projects/picamera_temp_humidity $ sudo npm install raspicam cloudinary node-dht-sensor node-dweetio
npm WARN install Refusing to install node-dht-sensor as a dependency of itself

> ws@0.4.32 install /usr/share/adafruit/webide/repositories/my-pi-projects/picamera_temp_humidity/node_modules/node-dweetio/node_modules/socket.io-client/node_modules/ws
> (node-gyp rebuild 2> builderror.log) || (exit 0)

make: Entering directory '/usr/share/adafruit/webide/repositories/my-pi-projects/picamera_temp_humidity/node_modules/node-dweetio/node_modules/socket.io-client/node_modules/ws/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node: Finished
  COPY Release/bufferutil.node
  CXX(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  SOLINK_MODULE(target) Release/obj.target/validation.node: Finished
  COPY Release/validation.node
make: Leaving directory '/usr/share/adafruit/webide/repositories/my-pi-projects/picamera_temp_humidity/node_modules/node-dweetio/node_modules/socket.io-client/node_modules/ws/build'
cloudinary@1.4.1 node_modules/cloudinary
├── q@1.4.1
└── lodash@3.10.1

raspicam@0.2.13 node_modules/raspicam
└── lodash@4.15.0

node-dweetio@0.0.11 node_modules/node-dweetio
├── request@2.33.0 (aws-sign2@0.5.0, forever-agent@0.5.2, qs@0.6.6, tunnel-agent@0.3.0, oauth-sign@0.3.0, json-stringify-safe@5.0.1, mime@1.2.11, node-uuid@1.4.7, form-data@0.1.4, tough-cookie@2.3.1, http-signature@0.10.1, hawk@1.0.0)
└── socket.io-client@0.9.16 (xmlhttprequest@1.4.2, uglify-js@1.2.5, active-x-obfuscator@0.0.1, ws@0.4.32)

The only error I got was running this:
$ wget http://node-arm.herokuapp.com/node_archive_armhf.deb
$ sudo dpkg -i node_archive_armhf.deb

Any ideas?