momenso/node-dht-sensor #27: Not getting readings

jamms, 2015-10-25:
Having read previous issues (most helpful was AuspXeu's post about node-gyp required to be installed) and spending a lot of time on this, I seem to have hit a brick wall! 

Output of ./check-bcm2835: 
gyp info ok response.

However, sudo node test.js:
Temperature: 0.0C, humidity: 0.0%, valid: true, errors: 1
Temperature: 0.0C, humidity: 0.0%, valid: true, errors: 0
Temperature: 0.0C, humidity: 0.0%, valid: true, errors: 0
Temperature: 0.0C, humidity: 0.0%, valid: true, errors: 0
etc

I know the Sensor is working (AM2302) as running a Python script gives me:

$ sudo ./AdafruitDHT.py 2302 4
Temp=22.8*  Humidity=57.6%

Node.js version: v0.12.6
NPM version: 2.11.2
bcm2835 Library version is 1.46
Pi Version: 2
Raspian: (Jessie)
node-dat-sensor: 0.0.8

I'm not a noob to Linux, but I am not that much of a whizz either, so could be missing something obvious here, but what?

TIA,

Julian

momenso, 2015-10-25:
If you're using ```test.js``` as is, it expects the sensor to be connected to pin **17** and it seems that you have your sensor connected to pin **4** by looking at your python script arguments.
It should work if you either connect the sensor to pin **17** or change the ```initialize``` function on test.js to read the sensor from pin **4**, like so:
```javascript
initialize: function() {
    this.totalReads = 0;
    return sensorLib.initialize(22, 4);
  },
```
Hope this helps.

jamms, 2015-10-25:
Doh! Do, you know, you are bang to rights on this one, I had (11, 4) specified, not sure why.. Thanks very much for your help David.

Hopefully, that will be it sorted from now on in!

Julian

momenso, 2015-10-26:
Great! Glad it worked.

closed 2015-12-02 by jamms