momenso/node-dht-sensor #28: Read once, then stop

jamms, 2015-12-02:
David,
Sorry, one more question, but is there a way you can show me of stopping the test.js or test-spec.js from running after providing the readings? I just want to take a reading, and then the script to stop running (either on success or error)...

TIA,

Jamms

momenso, 2015-12-02:
Yes, absolutely. Take any of the examples and remove the setTimeout and there you go. The setTimeout is there only so it will read again and again...

In the code below, I took the first example and commented out the setTimeout, as an example.
```javascript
var sensorLib = require('node-dht-sensor');

var sensor = {
    initialize: function () {
        return sensorLib.initialize(22, 4);
    },
    read: function () {
        var readout = sensorLib.read();
        console.log('Temperature: ' + readout.temperature.toFixed(2) + 'C, ' +
            'humidity: ' + readout.humidity.toFixed(2) + '%');
        //setTimeout(function () {
        //    sensor.read();
        //}, 2000);
    }
};

if (sensor.initialize()) {
    sensor.read();
} else {
    console.warn('Failed to initialize sensor');
}
```

jamms, 2015-12-05:
Well, that did the trick. I did try doing exactly this some time ago, without success, but it had been a long night :)

Thanks David, that has been very helpful to me.

closed 2015-12-05 by jamms