momenso/node-dht-sensor #31: Raspberry Pi 1 Model B

jnovack, 2015-12-12:
I have a number of pis all over, and I found the following so I wanted it to be documented somewhere.

Raspberry Pi 1 Model B
Raspbian "Wheezy"
Linux 3.10.21+ #604 PREEMPT Mon Dec 2 20:17:09 GMT 2013 armv6l GNU/Linux
Node v0.10.17
node-dht-sensor@0.0.6

rtclauss, 2015-12-27:
I've found that if you want to use this module with Node 4.x+ you need to update GCC on the RPi to at least 4.8.  I installed the module after following [this guide](https://somewideopenspace.wordpress.com/2014/02/28/gcc-4-8-on-raspberry-pi-wheezy/) prior to installing via npm.

jnovack, 2015-12-28:
Raspbian Jessie works out of the box, you only need to `update`/`upgrade` on Wheezy.

closed 2016-02-11 by momenso