momenso/node-dht-sensor #33: not installing node module

karthikbnd, 2016-02-08:
reply@raspberry:~/node_modules$ sudo npm install node-dht-sensor
/
> node-dht-sensor@0.0.11 preinstall /home/reply/node_modules/node-dht-sensor
> ./preinst

Library bcm2835 found.
Installing Native Abstractions for Node.js (NAN)...
npm WARN package.json node-dht-sensor@0.0.11 license should be a valid SPDX license expression
nan@2.2.0 node_modules/nan
Rebuiling addon...
make: Entering directory '/home/reply/node_modules/node-dht-sensor/build'
  CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o
  SOLINK_MODULE(target) Release/obj.target/node_dht_sensor.node
/usr/bin/ld: /usr/local/lib/libbcm2835.a(bcm2835.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libbcm2835.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
node_dht_sensor.target.mk:115: recipe for target 'Release/obj.target/node_dht_sensor.node' failed
make: *** [Release/obj.target/node_dht_sensor.node] Error 1
make: Leaving directory '/home/reply/node_modules/node-dht-sensor/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 3.18.7-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/reply/node_modules/node-dht-sensor
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
npm ERR! Linux 3.18.7-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "node-dht-sensor"
npm ERR! node v4.2.1
npm ERR! npm  v2.14.7
npm ERR! code ELIFECYCLE

npm ERR! node-dht-sensor@0.0.11 preinstall: `./preinst`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-dht-sensor@0.0.11 preinstall script './preinst'.
npm ERR! This is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./preinst
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/reply/node_modules/npm-debug.log

momenso, 2016-02-08:
As suggested in your logs, try re-compiling the BCM2835 library with '-fPIC'.
For example, go to the folder where you extracted the library package, and run the following commands:
```bash
$ ./configure
$ make clear
$ make CFLAGS='-g -O2 -fPIC'
$ sudo make check
$ sudo make install
```
Then try installing node-dht-sensor again.

InnovaGroup, 2016-03-09:
I had already done that but I have the same problem , what can be the solution ?

momenso, 2016-03-10:
If you recompiled the lib with -fPIC and it still fails with the same error then you might not have the correct lib installed on your system. I'd suggest you try following again the instructions and making sure every steps completes successfully.

closed 2016-08-14 by momenso