momenso/node-dht-sensor #34: Install Errors - gpy ERR!

mrkd, 2016-03-01:
```
Linux 4.1.13+ #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015 armv6l GNU/Linux
```

```
$ node --version
v5.7.0
```

```
$ npm --version
3.6.0
```

Output
```
node_dht_sensor.target.mk:88: recipe for target 'Release/obj.target/node_dht_sensor/node-dht-sensor.o' failed
make: *** [Release/obj.target/node_dht_sensor/node-dht-sensor.o] Error 1
make: Leaving directory '/home/pi/node-temperature-logger/node_modules/.staging/node-dht-sensor-07ec9ac4/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v5.7.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:100:13)
gyp ERR! stack     at ChildProcess.emit (events.js:185:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.1.13+
gyp ERR! command "/home/pi/.nvm/versions/node/v5.7.0/bin/node" "/home/pi/.nvm/versions/node/v5.7.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node-temperature-logger/node_modules/.staging/node-dht-sensor-07ec9ac4
gyp ERR! node -v v5.7.0
gyp ERR! node-gyp -v v3.2.1
gyp ERR! not ok 
node-temperature-logger@0.0.0 /home/pi/node-temperature-logger
└─┬ mysql@2.10.2 
  ├── bignumber.js@2.1.4 
  └─┬ readable-stream@1.1.13 
    ├── core-util-is@1.0.2 
    ├── inherits@2.0.1 
    ├── isarray@0.0.1 
    └── string_decoder@0.10.31 

npm WARN node-temperature-logger@0.0.0 No description
npm WARN node-temperature-logger@0.0.0 No repository field.
npm ERR! Linux 4.1.13+
npm ERR! argv "/home/pi/.nvm/versions/node/v5.7.0/bin/node" "/home/pi/.nvm/versions/node/v5.7.0/bin/npm" "install"
npm ERR! node v5.7.0
npm ERR! npm  v3.6.0
npm ERR! code ELIFECYCLE

npm ERR! node-dht-sensor@0.0.11 preinstall: `./preinst`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-dht-sensor@0.0.11 preinstall script './preinst'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./preinst
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-dht-sensor
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/node-temperature-logger/npm-debug.log
```

mrkd, 2016-03-01:
I do have the

fiws, 2016-03-10:
more or less the same issue with 0.0.11 – worked fine with 0.0.8:

```
> node-dht-sensor@0.0.11 preinstall /tmp/node_modules/.staging/node-dht-sensor-7ad49927
> ./preinst

Library bcm2835 found.
Installing Native Abstractions for Node.js (NAN)...
npm ERR! Linux 4.1.18-1-ARCH
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "nan"
npm ERR! node v5.8.0
npm ERR! npm  v3.8.0

npm ERR! Cannot read property 'target' of null
npm ERR! 
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/node_modules/.staging/node-dht-sensor-7ad49927/npm-debug.log
Rebuiling addon...
module.js:341
    throw err;
    ^

Error: Cannot find module 'nan'
    at Function.Module._resolveFilename (module.js:339:15)
    at Function.Module._load (module.js:290:25)
    at Module.require (module.js:367:17)
    at require (internal/module.js:16:19)
    at [eval]:1:1
    at Object.exports.runInThisContext (vm.js:54:17)
    at Object.<anonymous> ([eval]-wrapper:6:22)
    at Module._compile (module.js:413:34)
    at node.js:544:27
    at _combinedTickCallback (node.js:370:9)
gyp: Call to 'node -e "require('nan')"' returned exit status 1 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:305:16)
gyp ERR! stack     at emitTwo (events.js:100:13)
gyp ERR! stack     at ChildProcess.emit (events.js:185:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Linux 4.1.18-1-ARCH
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /tmp/node_modules/.staging/node-dht-sensor-7ad49927
gyp ERR! node -v v5.8.0
gyp ERR! node-gyp -v v3.3.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/tmp/package.json'
npm WARN tmp No description
npm WARN tmp No repository field.
npm WARN tmp No README data
npm WARN tmp No license field.
npm ERR! Linux 4.1.18-1-ARCH
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "node-dht-sensor"
npm ERR! node v5.8.0
npm ERR! npm  v3.8.0
npm ERR! code ELIFECYCLE

npm ERR! node-dht-sensor@0.0.11 preinstall: `./preinst`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-dht-sensor@0.0.11 preinstall script './preinst'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./preinst
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-dht-sensor
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/npm-debug.log
```

momenso, 2016-03-10:
@fiws not really the same the issue. Yours is failing when installing nan module. You could try ```npm cache clean```and/or try installing nan manually to see if it works: ```npm install --save nan```

fiws, 2016-03-11:
@momenso cleaning the cache changed nothing. Installed nan manually solved it (I think):
```
> node-dht-sensor@0.0.11 preinstall /tmp/node_modules/.staging/node-dht-sensor-7ad49927
> ./preinst

Library bcm2835 found.
Installing Native Abstractions for Node.js (NAN)...
npm ERR! Linux 4.1.18-1-ARCH
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "nan"
npm ERR! node v5.8.0
npm ERR! npm  v3.8.0

npm ERR! Cannot read property 'target' of null
npm ERR! 
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/node_modules/.staging/node-dht-sensor-7ad49927/npm-debug.log
Rebuiling addon...
make: Entering directory '/tmp/node_modules/.staging/node-dht-sensor-7ad49927/build'
  CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o
  SOLINK_MODULE(target) Release/obj.target/node_dht_sensor.node
  COPY Release/node_dht_sensor.node
make: Leaving directory '/tmp/node_modules/.staging/node-dht-sensor-7ad49927/build'
/tmp
+-- nan@2.2.0 
`-- node-dht-sensor@0.0.11 
```

closed 2016-08-14 by momenso