momenso/node-dht-sensor #35: issue : ACTION Regenerating Makefile

mehamindran, 2016-03-07:
make: Warning: File '../binding.gyp' has modification time 6157163 s in the future
  ACTION Regenerating Makefile
make: Warning: File '../binding.gyp' has modification time 6157163 s in the future
  ACTION Regenerating Makefile

This is happening When I do "npm install node-dht-sensor",
This is not stoping at any point,It's infinite loop.

momenso, 2016-03-10:
This most likely related to your system having date/time badly configured.

closed 2016-05-15 by momenso

umar-muneer, 2020-09-28:
> This most likely related to your system having date/time badly configured.

A four-year-old comment saved at least an hour's effort. Thanks 👍

rdwnsjjd, 2023-05-02:
> > This most likely related to your system having date/time badly configured.
> 
> A four-year-old comment saved at least an hour's effort. Thanks 👍

and for me after two years from this comment!
thanks again!