momenso/node-dht-sensor #36: Could not install module

zerr0s, 2016-04-06:
Hi,

Forgive me for my bad english. I have an issue while trying to install the module. This module wasn't installed before.

make: Entering directory '/home/spraybot/sprayBot/node_modules/.staging/node-dht-sensor-57b2f592/build'
  CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o
  SOLINK_MODULE(target) Release/obj.target/node_dht_sensor.node
  COPY Release/node_dht_sensor.node
make: Leaving directory '/home/spraybot/sprayBot/node_modules/.staging/node-dht-sensor-57b2f592/build'
npm ERR! Linux 4.1.20-v7+
npm ERR! argv "/opt/node-v5.9.1-linux-armv7l/bin/node" "/usr/bin/npm" "install" "node-dht-sensor" "--save"
npm ERR! node v5.9.1
npm ERR! npm  v3.7.3
npm ERR! path /home/spraybot/sprayBot/node_modules/node-dht-sensor
npm ERR! code ENOTEMPTY
npm ERR! errno -39
npm ERR! syscall rename

npm ERR! ENOTEMPTY: directory not empty, rename '/home/spraybot/sprayBot/node_modules/node-dht-sensor' -> '/home/spraybot/sprayBot/node_modules/.node-dht-sensor.DELETE'
npm ERR! 
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /home/spraybot/sprayBot/npm-debug.log

Is there a way to setup the module ?

Thanks.

momenso, 2016-04-06:
Looks like something wrong with your project directory structure.
Can you remove ```/home/spraybot/sprayBot/node_modules/``` and try installing the dependencies again?

zerr0s, 2016-04-07:
Hi. I've tried to remove the node-modules directory and reinstall all modules but same issue. I finally tried to install it globally in root user. It works.

Thanks for te answer.

Khadgar, 2016-05-15:
Hi!

I am trying to use your library but I am facing a problem during the Rebuild addon part of the npm install.

Setup:

root@danipi:~/DEV/WEB/myproject# **node** --version
**v0.10.29**
root@danipi:~/DEV/WEB/myproject# **npm** --version
**1.4.21**
root@danipi:~/DEV/WEB/myproject# **node-gyp** --version
**v0.12.2**

I've successfully installed the [**bcm2835**](http://www.airspayce.com/mikem/bcm2835/) lib version: **1.50**

Error log:

```
root@danipi:~/DEV/WEB/myproject# npm install node-dht-sensor
|
> node-dht-sensor@0.0.11 preinstall /root/DEV/WEB/myproject/node_modules/node-dht-sensor
> ./preinst

Library bcm2835 found.
Installing Native Abstractions for Node.js (NAN)...
nan@2.3.3 node_modules/nan
Rebuiling addon...
make: Entering directory '/root/DEV/WEB/myproject/node_modules/node-dht-sensor/build'
  CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o
In file included from ../node-dht-sensor.cpp:200:0:
../node_modules/nan/nan.h:330:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
node_dht_sensor.target.mk:82: recipe for target 'Release/obj.target/node_dht_sensor/node-dht-sensor.o' failed
make: *** [Release/obj.target/node_dht_sensor/node-dht-sensor.o] Error 1
make: Leaving directory '/root/DEV/WEB/myproject/node_modules/node-dht-sensor/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.1.19+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /root/DEV/WEB/myproject/node_modules/node-dht-sensor
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR! node-dht-sensor@0.0.11 preinstall: `./preinst`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-dht-sensor@0.0.11 preinstall script.
npm ERR! This is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./preinst
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.19+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "node-dht-sensor"
npm ERR! cwd /root/DEV/WEB/myproject
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /root/DEV/WEB/myproject/npm-debug.log
npm ERR! not ok code 0
```
I am running everything as a root user on a RaspberryPI 1

Thanks for helping,
Daniel

momenso, 2016-05-15:
Do you need to use node v0.10 and npm 1.4?
My recommendation is to get rid of such ancient versions and get a more recent version of npm, node and node-gyp. The issue you describe doesn't affect more modern versions such as: ```npm v2.14```, ```node v4.2.1``` and ```node-gyp v3.3.1```.

Khadgar, 2016-05-16:
Hi!

Yes, the problem was the old packages. The debian-packaged node is missing the REPLACE_INVALID_UTF8 patch from v0.10.29.

Now my setup (The latest nodejs with ARMv6 support because of RaspberryPi 1 B):

root@DietPi:~/DEV/WEB/myproject# **npm** --version
**2.15.1**
root@DietPi:~/DEV/WEB/myproject# **node** --version
**v4.4.4**
root@DietPi:~/DEV/WEB/myproject# **node-gyp** --version
**v3.3.1**

**bcm2835** lib version: **1.50**

Everything works thank you for the quick answer :)

Best regards,
Daniel

closed 2016-05-16 by momenso