momenso/node-dht-sensor #42: rpi 3 return values always returning 0.0

emrekilic, 2016-05-26:
I am using dht11 with rpi3. 

var sensorLib = require('node-dht-sensor');

var sensor = {
    initialize: function () {
        return sensorLib.initialize(11, 4);
    },
    read: function () {
        var readout = sensorLib.read();
        console.log('Temperature: ' + readout.temperature.toFixed(2) + 'C, ' +
            'humidity: ' + readout.humidity.toFixed(2) + '%');
        setTimeout(function () {
            sensor.read();
        }, 2000);
    }
};

if (sensor.initialize()) {
    sensor.read();
} else {
    console.warn('Failed to initialize sensor');
}

I am using this example code. Dht11's data pin connected to gpio4. bcm2835 library also installed. But return values always 0.0. I am sure that connection is true, because i can get temperature with c code.

emrekilic, 2016-05-26:
i have also nrf24 module connected into rpi 3. I think it also using bcm2835 library. If i not use it, now i can get temperature. But i didnt understand why they are not work together?

momenso, 2016-05-27:
Are you running node with sudo?

cgalavis, 2016-06-06:
Was this resolved? I am in the process of rewriting a python based data logger in Node and was planning to use this module to read from an am2302 sensor, the software runs in a rpi 3.

Thanks!

emrekilic, 2016-06-06:
Yes. Its working with rpi 3. You can use it. My problem is not related with this library.

closed 2016-08-14 by momenso