momenso/node-dht-sensor #45: Strange doubling of Temperature

fletchto99, 2016-08-14:
Hey!

I've recently set up an application to read the temperature in my room and ocasionally it will double the temperature value. I would say this occurs about once every 12 hours or so.

I'm using the DHT22 sensor with a status LED. Here's an image of the issue I'm facing, notice the temperature.

[image]

The code I'm using to read out the data appears to be inline with the example.
````
const sensorLib = require('node-dht-sensor');
const config = require('../config.json').sensor;
const LED = require('./LED');

sensorLib.initialize(22, config.gpio);

module.exports =  {
    read() {
        LED.enable();
        var readout = sensorLib.read();
        LED.disable();
        return readout;
    }
};
````

Any ideas as to why this might be occuring?

momenso, 2016-08-14:
I haven't came across this yet. I am running a similar code to yours to see if I can reproduce it.
In the mean time I recommend you upgrade to the latest ```0.0.13``` version that addresses a bug with the BCM library.

fletchto99, 2016-08-14:
I'll do that. I might already be on 0.0.13 since I just added the
temperature sensor 3 days ago. I'll double check though!

Thanks.

fletchto99, 2016-08-15:
I've got the Rpi 2 running Raspbian & Node v6.3.1

I've followed this guide to setup the sensor: https://www.youtube.com/watch?v=IHTnU1T8ETk

fletchto99, 2016-08-15:
Hmm something strange is definitely
[image]
 going on. It has now recorded a temperature of 99.9?

momenso, 2016-08-16:
I suspect it has to do with the protocol timing and the decoding but I need to investigate this further.
I did reproduce the doubling... and you're right it doesn't happen very often. I am running again with a trace this time to see it provides any clues on what might be causing it.
For now I am updating the decoding algorithm with more conservative timings that should result in more reliable readings overall.

fletchto99, 2016-08-16:
Perfect thank you very much for looking into the issue. I'll update to the latest version now and see if that makes a difference.

fletchto99, 2016-08-16:
I'm not sure if this helps but I'm getting alot of these types of logs:

````

temperature-5 (out): Unexpected data: j=39: 56 != 2 + 52 + 2 + 2
temperature-5 (out): Unexpected data: j=40: 55 != 2 + 49 + 1 + 1
temperature-5 (out): Unexpected data: j=39: 55 != 2 + 51 + 2 + 2
temperature-5 (out): Unexpected data: j=39: 57 != 4 + 106 + 2 + 2
temperature-5 (out): Unexpected data: j=39: 57 != 4 + 106 + 2 + 2
temperature-5 (out): Unexpected data: j=39: 57 != 2 + 106 + 2 + 2
temperature-5 (out): Unexpected data: j=40: 57 != 2 + 37 + 1 + 1
temperature-5 (out): Unexpected data: j=39: 57 != 2 + 58 + 2 + 2
temperature-5 (out): Unexpected data: j=39: 57 != 2 + 52 + 2 + 2
temperature-5 (out): Unexpected data: j=39: 58 != 2 + 54 + 2 + 2
temperature-5 (out): Unexpected data: j=39: 59 != 2 + 54 + 2 + 4
temperature-5 (out): Unexpected data: j=39: 57 != 4 + 106 + 2 + 2
temperature-5 (out): Unexpected data: j=39: 58 != 2 + 53 + 1 + 4
temperature-5 (out): Unexpected data: j=40: 58 != 2 + 49 + 1 + 2
temperature-5 (out): Unexpected data: j=39: 56 != 4 + 104 + 2 + 2
temperature-5 (out): Unexpected data: j=40: 56 != 0 + 52 + 1 + 1
temperature-5 (out): Unexpected data: j=40: 56 != 2 + 20 + 1 + 1
temperature-5 (out): Unexpected data: j=39: 58 != 2 + 54 + 2 + 4
temperature-5 (out): Unexpected data: j=40: 56 != 0 + 52 + 1 + 1
````

momenso, 2016-08-16:
Yes, that's helpful. It contains protocol state when CRC validation fails.

fletchto99, 2016-08-16:
Ok perfect. I've also noticed that it is now recording other strange temperatures, such as 0.02 degrees Celsius. Also I noticed there are some inconsistencies in the attempts at checking the temperature now, see the chart:
[image]
 
For a realtime preview see here: https://temperature.fletchto99.com/

momenso, 2016-08-17:
There is a new update that you should try (```node-dht-sensor@0.0.17```). I have been running it for the past few hours without any glitches. It's probably too soon to say it's the fix for this issue but it's looking promising. Let me know if you have a chance to test it too.

fletchto99, 2016-08-17:
I will try it out and report back. I'll keep an eye on the logs as well to see if anything pops up.

fletchto99, 2016-08-17:
Unfortunately version 0.0.17 didn't seem to fix the issue.

[image]

fletchto99, 2016-08-25:
Seems to still be happening with ver 0.0.18. I've posted the code for it all now on my github: https://github.com/fletchto99/termperature.fletchto99.com 

I know the package.json states 0.0.12 however the deployed version is running 0.0.18

momenso, 2016-09-13:
This problem is caused by loss of a few bits when reading from the sensor. Although there's a CRC in the protocol to help mitigate communication problems such as this, in some extreme cases CRC is not able to detect the error and you get those bad readings. :boom:

To try and mitigate this problem there's now additional checking on the data received from the sensor  to ensure it not only matches the expected CRC but also contains the exact number of bits for proper decoding.

This is now implemented in version ```0.0.23```. I have been running it for the past 24 hours without any issues whereas with previous versions the issue would already have popped up for sure.

If you have a chance, try updating it and let us know if you still see the problem.

fletchto99, 2016-09-13:
Huh strange that the CRC check doesn't detect this! I'll update now and report back.

fletchto99, 2016-09-14:
Hi Momenso. Unfortunately I don't think this has fully fixed the issue.

I still have an outlier at 48.2 degrees after updating.

[image]

momenso, 2016-09-16:
I have been running a test case for this issue for a few days now without any issues. Are you sure you're running an updated version of the sensor library?
Here are a sample of the data gathered in the test. As you can see there's no "outlier". 
[image]
In the meantime I updated the library again to **node-dht-sensor@0.0.25** and I will continue tests and will update if I find anything.

fletchto99, 2016-09-17:
I'll test 0.0.25 now and see if it fixed it. Perhaps 23 actually fixed it, I feel like an idiot but I actually forgot to restart the process after installing 0.0.23. I've installed the latest version now and will report back in 24 hrs or so.

momenso, 2016-09-18:
No worries. It's running fine over here so far... but let me know how it goes for you.

fletchto99, 2016-09-19:
@momenso Woo I think it's fixed! Been running fine all weekend!

closed 2016-09-19 by fletchto99