momenso/node-dht-sensor #46: Refusing to install node-dht-sensor as a dependency of itself

techk9solutions, 2016-09-08:
I've followed all of your suggestions with the following two exceptions that errored out:

1.  Errors on running this: sudo dpkg -i node_latest_armhf.deb
  dpkg: regarding node_latest_armhf.deb containing node:
 nodejs-legacy conflicts with node
  node (version 4.2.1-1) is to be installed.
  node provides node and is to be installed.

dpkg: error processing archive node_latest_armhf.deb (--install):
 conflicting packages - not installing node
Errors were encountered while processing:
 node_latest_armhf.deb

Here is my overall error:
/usr/share/adafruit/webide/repositories/my-pi-projects/picamera_temp_humidity $ sudo npm install raspicam cloudinary node-dht-sensor node dweetio
npm WARN package.json node@0.0.0 No description
npm WARN package.json node@0.0.0 No repository field.
npm WARN package.json node@0.0.0 No README data
npm WARN install Refusing to install node-dht-sensor as a dependency of itself
npm WARN deprecated node@0.0.0: To update or install node, go to http://nodejs.org/
npm WARN engine cryptiles@2.0.5: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine hoek@2.16.3: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine boom@2.10.1: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
node@0.0.0 node_modules/node

cloudinary@1.4.1 node_modules/cloudinary
├── q@1.4.1
└── lodash@3.10.1

raspicam@0.2.13 node_modules/raspicam
└── lodash@4.15.0

dweetio@0.0.1 node_modules/dweetio
├── mime@1.3.4
├── bluebird@3.4.6
└── request@2.58.0 (caseless@0.10.0, aws-sign2@0.5.0, oauth-sign@0.8.2, forever-agent@0.6.1, tunnel-agent@0.4.3, stringstream@0.0.5, isstream@0.1.2, extend@2.0.1, json-stringify-safe@5.0.1, combined-stream@1.0.5, node-uuid@1.4.7, qs@3.1.0, mime-types@2.0.14, tough-cookie@2.3.1, http-signature@0.11.0, bl@0.9.5, hawk@2.3.1, har-validator@1.8.0, form-data@1.0.1)

Seems I'm getting close.  Is this related the node.js error earlier?  Any ideas?

momenso, 2016-09-08:
Here another way to get latest node.js up and running on your raspberry pi.

```bash
$ curl -sL https://deb.nodesource.com/setup_6.x | sudo -E bash -
$ sudo apt install -y nodejs
```

With that, you should have latest node and npm installed.

techk9solutions, 2016-09-09:
Installed 4.x and still getting "Refusing to install node-dht-sensor as a dependency of itself.

/home/webide/repositories/my-pi-projects/picamera_temp_humidity $ sudo npm install raspicam cloudinary node-dht-sensor node-dweetio
npm WARN install Refusing to install node-dht-sensor as a dependency of itself
-
> ws@0.4.32 install /usr/share/adafruit/webide/repositories/my-pi-projects/picamera_temp_humidity/node_modules/node-dweetio/node_modules/socket.io-client/node_modules/ws
> (node-gyp rebuild 2> builderror.log) || (exit 0)

make: Entering directory '/usr/share/adafruit/webide/repositories/my-pi-projects/picamera_temp_humidity/node_modules/node-dweetio/node_modules/socket.io-client/node_modules/ws/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
bufferutil.target.mk:96: recipe for target 'Release/obj.target/bufferutil/src/bufferutil.o' failed
make: Leaving directory '/usr/share/adafruit/webide/repositories/my-pi-projects/picamera_temp_humidity/node_modules/node-dweetio/node_modules/socket.io-client/node_modules/ws/build'
cloudinary@1.4.1 node_modules/cloudinary
├── q@1.4.1
└── lodash@3.10.1

raspicam@0.2.13 node_modules/raspicam
└── lodash@4.15.0

node-dweetio@0.0.11 node_modules/node-dweetio
├── request@2.33.0 (aws-sign2@0.5.0, forever-agent@0.5.2, tunnel-agent@0.3.0, qs@0.6.6, oauth-sign@0.3.0, mime@1.2.11, json-stringify-safe@5.0.1, tough-cookie@2.3.1, node-uuid@1.4.7, http-signature@0.10.1, form-data@0.1.4, hawk@1.0.0)
└── socket.io-client@0.9.16 (xmlhttprequest@1.4.2, uglify-js@1.2.5, active-x-obfuscator@0.0.1, ws@0.4.32)

Seem to be stuck on node-dht-sensor.

techk9solutions, 2016-09-09:
Additional - tried manual install of node-dht-sensor:  Here is the error:

/home/webide/repositories/my-pi-projects/picamera_temp_humidity# npm install node-dht-sensor
npm ERR! Linux 4.4.13-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "node-dht-sensor"
npm ERR! node v6.5.0
npm ERR! npm  v3.10.3
npm ERR! code ENOSELF

npm ERR! Refusing to install node-dht-sensor as a dependency of itself
npm ERR!
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /usr/share/adafruit/webide/repositories/my-pi-projects/picamera_temp_humidity/npm-debug.log
root@raspberrypi:/home/webide/repositories/my-pi-projects/picamera_temp_humidity# exit
exit

momenso, 2016-09-09:
By the looks of it, you have a ```package.json``` on your current directory with ```name: "node-dht-sensor"``` that conflicts with the package you're asking npm to install. If that's the case, you should rename it to something different.
Preferable, I recommend you start fresh in a new folder, and initialise your project with ```npm init``` where it will assist you on creating a proper ```package.json``` file - just make sure you don't name your own project "node-dht-sensor" or any of other package names you depend on.

closed 2016-09-16 by momenso