momenso/node-dht-sensor #47: Script output 0 temp 0 hum

Yoda49, 2016-09-16:
With logic analyzer i get this signal. Its ok, but after CRC byte - there are three more bytes with zero.
[image]

Init OK.
The script shows temperature 0 and pressure 0.

momenso, 2016-09-16:
Nice! I haven't connected a logic analyzer yet to check this. I need to get one of these! (-:

In the current version the algorithm will discard this reading as invalid and you should be able to confirm it by the ```isValid``` attribute of the readout as follows.
```javascript
var readout = sensorLib.read();
readout.isValid // should be false in this case
```
Are you able to confirm if this 24 extra bits are always present? Even if you try to read again the sensor?

Yoda49, 2016-09-16:
Yes, isValid = false, after reading sensor. Each time the sensor reading - at the end there are 24 null bits.

momenso, 2016-09-16:
I couldn't find any sensors that behave like this but I added a patch to ```node-dht-sensor@0.0.24``` that should allow for it to read the sensor data even if these extra bits are present.
Give it a try and let me know what happens.

Yoda49, 2016-09-16:
Now Avalid is TRUE, but script output temp: 1.00C and hum: 1.00% Its like on most significant byte of temp and hum. Problem in code.
[image]
[image]

momenso, 2016-09-16:
It looks like you didn't configure the sensor type to decode from a **DHT22** sensor (i.e. type=22).
Make sure you initialize the sensor using the `sensorLib.initialize` and specify both the sensor type and the GPIO pin accordingly.
For example, you can see In the [first example](https://github.com/momenso/node-dht-sensor#first-example) the library is initialized to read a **DHT22** sensor from GPIO `4.` Otherwise, the library will default to DHT11 and it will produce the result you're getting.

Yoda49, 2016-09-16:
I am sorry)  I forgot to change the sensor identifier from 11 to 22. Its work! Thanks!!!!

momenso, 2016-09-16:
:thumbsup:

closed 2016-09-16 by momenso