momenso/node-dht-sensor #49: sensorLib.read() is blocking

adifah, 2016-09-26:
Hi folks,

while playing around with my raspberry pi and some sensors/actors (dht11 / button / led) i observed, that the read() method seems to block the application in a synchronous manner (executing the read() method on my rpi b+ takes something from a few milliseconds up to 3 seconds).

My simple application reads the sensor data every 5 seconds, besides a led lights up if i press the button (using https://github.com/fivdi/pigpio#buttons-and-interrupt-handling). The problem is, if i press the button while the read() method gets executed nothing happens until the read operation is finished. Then the interrupt handler for the button gets called only once no matter how often i pressed the button in the meanwhile.

I'm not sure if an asynchronous/callback approach would solve this problem, but missing sensor inputs might become a problem in some scenarios.

Thanks in advance!

momenso, 2016-09-30:
You can try to use the new asynchronous read function available in ```node-dht-sensor@0.0.25```.
For example, the following code will read a DHT22 sensor connected to GPIO port 4 and will not block until the sensor responds back.

```javascript
var sensor = require('node-dht-sensor');

sensor.read(22, 4, function(err, temperature, humidity) {
    if (err) {
        console.warn('' + err);
    } else {
        console.log('temp: ' + temperature.toFixed(1) + '°C, ' +
            'humidity: ' + humidity.toFixed(1) + '%'
        );
    }
});
```

adifah, 2016-10-01:
great, thanks a lot for your affort!
I will take a look in the next days

closed 2017-01-30 by momenso