momenso/node-dht-sensor #50: Module always returns 0 with DHT11 and rPi 3

vannut, 2016-10-01:
I've got a DHT11 connected to GPIO26 on my Raspberry Pi 3.
When I read this DHT11 with a simple python script it returns the correct temp and hum. So the DHT functions and is giving good values.

But when using nodejs v6.7.0 and the 'first example' script both values are 0.

Any idea why?

momenso, 2016-10-01:
Most likely it is because sensor type and/or GPIO pin is not correctly set. In your case, sensor type should be 11 and GPIO pin number should be according to BCM2835 convention. More info [here](https://www.raspberrypi.org/documentation/usage/gpio/).

vannut, 2016-10-01:
This is the script used:
var sensor = require('node-dht-sensor');

```
sensor.read(11, 26, function(err, temperature, humidity) {
    if (!err) {
        console.log(temperature);
        console.log('temp: ' + temperature.toFixed(1) + '°C, ' +
            'humidity: ' + humidity.toFixed(1) + '%'
        );
    }
});
```
So that should be fine right?

adifah, 2016-10-01:
I had the same problem when i used the 3.3V pin and a 10kΩ resistor.
Try 5V (should be within specifications) instead, that solved the probem for me.

momenso, 2016-10-01:
Why GPIO pin 26? Are you using some kind of expansion port?
Could you try connecting to something like, directly to pin 4?
[image]

vannut, 2016-10-02:
I'm indeed using a kind of shield with another pcb on top of that. It also has a footprint for a DHT sensor. 
On the shield there are also some groove connectors, which is connected to: GPIO13 and GPIO19; both of them don't work in nodejs. Although both work in python.

momenso, 2016-10-02:
If you can read the sensor via a python script it should be possible to read via node.js too. It most likely is due to different GPIO pin numbering schemes. Some libraries map physical GPIO differently.

eucuepo, 2016-10-02:
I am having the same problem on raspberry pi 3. Tried 5 volts with a 10k resistor, still gives 0/0

vannut, 2016-10-03:
@eucuepo which port(s) did you try?
There might be an issue in addressing the 'upper' GPIO's.. (non rpi A/B)

eucuepo, 2016-10-04:
Finally got it working on GPIO 13:
- using 5V input
- connecting the 5V to the data pin with a 10K resistor
The key for me was using the 5V directly from the raspberry to the resistor, not using the same line that goes to the DHT. I'm don't know much about electronics, should be the same but it was not working for me when using the same line

closed 2016-10-08 by momenso