momenso/node-dht-sensor #51: Cannot install version 0.0.25 or 0.0.26 on jessie 4.4.22 - needs c++ 2011 support ?

gundrygs, 2016-10-06:
I am able to install version 0.0.24 OK, but not 0.0.25 or 0.0.26 - I get the following fatal error when using npm install node-dht-sensor@0.0.26
                           
  CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o                                                    
In file included from /usr/include/c++/4.9/mutex:35:0,                                                                       
from ../node-dht sensor.cpp:4:                                                                                                                        
/usr/include/c++/4.9/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.

I am attempting this on a Pi3 with Raspbian Jessie 4.4.22

momenso, 2016-10-08:
Yes, C++11 is required. 
Are you still seeing this issue on ```node-dht-sensor@0.0.27```?

gundrygs, 2016-10-08:
Thanks for responding - I will try this on Monday, but I'm pretty confident your addition of the cflags directive to binding.gyp will do the trick.

Please could you also look at the implementation of ReadSpec in node-dht-sensor.cpp - it uses a do .. while construct at lines 108 to 111, crucially (for us) without a sleep(1) between retries, whereas ReadPreset uses a while (true) construct at lines 127 to 132 with the sleep(1) statement - is there a reason for this difference or is it just an oversight in your recent refactoring ?

Please let me take the opportunity to thank you for your work on this library.

momenso, 2016-10-08:
There should be no difference really. The ```sleep``` there is important because a cooldown period is required between sensor reads. Otherwise, if reading the sensor fails on the first attempt and we try to read it immediately again then it will keep on failing.

I am guessing you need to read the sensor very quickly, right? If that's the case, I have included a new method on the library that allows you to effectively have no extra delay when reading the sensor. **So, you can disable that to have no extra delays but then you have to make sure to handle reading errors yourself**.

By default, the library will attempt to query the sensor up to three (3) times before returning an invalid readout. Overall this is a very reasonable setting which results in good readouts in most situations. Every time the library needs to read the sensor again due to a failure, it will apply an 450ms delay (cooldown periord, whereas before it was 1s). So, if you need to have a response in less than that, you can either ```setMaxRetries(0)``` or ```initialize(<type>, <gpioPin>, 0)``` the sensor which will effectively configure the lib to return with a readout in less than 25ms (typically).

This is an example using asynchronous call to read the sensor:
```javascript
var sensor = require('node-dht-sensor');

// disable auto retry feature, default is 3
sensor.setMaxRetries(0); 

sensor.read(22, 4, function(err, temperature, humidity) {
    if (err) {
      console.warn(err);
      // TODO: do something about it
    } else {
      console.log('temperature: ' + temperature.toFixed(1) + '°C, ' +
        'humidity: ' + humidity.toFixed(1) + '%');
    }
});
```
Or if you don't want to deal with callbacks, you can use the synchronous call by initializing the sensor and just reading when needed.
```javascript
var sensor = require('node-dht-sensor');

// initialize the sensor with sensor type, GPIO pin and set maxRetries to zero
if (!sensor.initialize(22, 4, 0)) {
  console.warn('failed to initialize sensor');
  return;
}

var readout = sensor.read();
if (!readout.isValid) {
    console.warn('failed to read sensor');
    // TODO: do something about it
} else {
    console.log('temp: ' + readout.temperature.toFixed(1) + '°C, ' +
        'humidity: ' + readout.humidity.toFixed(1) + '%');
}
```
Third option, would be to just call the ```setMaxRetries``` method and use the synchronous read call with specific sensor type and GPIO pin.
```javascript
var sensor = require('node-dht-sensor');

// disable auto retry feature, default is 3
sensor.setMaxRetries(0);
var readout = sensor.read(22, 4);
if (!readout.isValid) {
    console.warn('failed to read sensor');
    // TODO: do something about it
} else {
    console.log('temp: ' + readout.temperature.toFixed(1) + '°C, ' +
        'humidity: ' + readout.humidity.toFixed(1) + '%');
}
```
In the examples above, I am reading a *DHT22* sensor on GPIO pin 4.

gundrygs, 2016-10-09:
Actually, we are using ReadSpec to read multiple DHT22 sensors on different GPIO pins attached to 50m runs of Cat5 cable, so retries are quite common and we wanted the 1 second back off ...
However, your solution is even better, as we now have complete control of the retries, how many and how long to wait between - thank you !

gundrygs, 2016-10-10:
All working well here - although clearly the removal of a separate readSpec() function call [now using read() with 2 parameters] - is a breaking API change for existing users - is this what you intended ?
I am happy for you to close the original issue as this is resolved by the addition of the cflags directive to binding.gyp.

momenso, 2016-10-10:
Yes, that's the new API moving forward - hopefully it's small enough to cause not much trouble :grimacing: and a cleaner API.

closed 2016-10-10 by momenso