momenso/node-dht-sensor #52: TypeError: sensorLib.readSpec is not a function

ceburge, 2016-10-07:
Trying to use DHT22 in Node Red. Everything seems to have installed fine but keep getting the error in the title.
Nodr Rec version 14.6
Node version 4.6.0
npm version 2.15.9

Flow:
[{"id":"40588b83.b5ccb4","type":"rpi-dht22","z":"ec482b12.18a198","name":"","topic":"rpi-dht22","dht":22,"pintype":"0","pin":"14","x":695,"y":237,"wires":[["cd8fe79f.9e7c78"]]},{"id":"5b0e8fec.3b0c6","type":"inject","z":"ec482b12.18a198","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"x":384,"y":236,"wires":[["40588b83.b5ccb4"]]},{"id":"cd8fe79f.9e7c78","type":"debug","z":"ec482b12.18a198","name":"","active":true,"console":"false","complete":"false","x":969,"y":237,"wires":[]}]

momenso, 2016-10-08:
I've submitted a pull request to fix this issue with *node-red-contrib-dht-sensor@0.0.17* but if you want to try the fix now you can try installing this fork:
```bash
$ cd ~/.node-red
$ npm install git+https://github.com/momenso/node-red-contrib-dht-sensor.git
```
This should make your existing flow work as expected.

ceburge, 2016-10-10:
Works great. Thank you.

closed 2016-10-10 by ceburge