momenso/node-dht-sensor #53: Sometimes script gives double value of temperature and humidity.

Yoda49, 2016-10-13:
Sensor AM2301 (DHT21).
Began to see that sometimes temperature and humidity have the doubled value in comparison with last reading.  
Example: temp 25 С, hm 40%.  Next reading i get 50 С and 80 %. It is similar to left bit shift. 
This bug happens quite seldom and I can't catch it by means of the logic analyzer, but it is visible on the diagram which is built of the written data file.

momenso, 2016-10-13:
Can you confirm what version of ```node-dht-sensor``` are you using? 
This looks the same as #45.

Yoda49, 2016-10-13:
"version": "0.0.24"

momenso, 2016-10-13:
Ok. that explains the problem. Better yet you should update to the latest version - currently ```node-dht-sensor@0.0.28``` which includes improvements on the unified interface as well.

Yoda49, 2016-10-14:
I update version to 0.0.28 the problem still exists.
[image]

momenso, 2016-10-14:
Hi @Yoda49, this sure looks like the bug from issue #45 still. 

I have been running a test to see if I could get this error again but it didn't happen yet. And by the looking at your logs it seems that it didn't take much time for it to manifest, right?

Would it be possible that you did install the latest version but did not restart the node.js server? If so, that would explain why you still see the problem. So, just to make sure, please confirm you have node-dht-sensor@0.0.28 and then restart your server.

Yoda49, 2016-10-17:
"version": "0.0.28", - it from "manifest .json" and rebooted the server many times. On these days off caught this error two times again. The first time temperature in humidity. And second time only temperature.
[image]

Yoda49, 2016-10-20:
Again doubling. of data. Raspberry and nodejs-server were rebooted before many times
[image]

momenso, 2016-10-20:
I have been trying to reproduce this issue on the latest version with no success.

Sorry to insist on you using the latest version but just to make sure we are on the same page, can you try to ```npm list``` your project directory to double check the node-dht-sensor version?
You should get something like this:
```bash
pi@raspberrypi:~/myproject $ npm list
myproject@1.0.0 /home/pi/myproject
└─┬ node-dht-sensor@0.0.28
  └── nan@2.4.0
```

Yoda49, 2016-10-21:
[image]

Yoda49, 2016-10-27:
There are any thoughts about it?

momenso, 2016-10-27:
Can you try running the library on verbose mode? It should provide more information to help understand what's causing this value doubling...
You can install the ```node-dht-sensor``` in verbose mode as follows.
```bash
$ npm install node-dht-sensor --dht_verbose=true
```
When verbose mode is enabled, a log file (**dht-sensor.log**) will be generated on the directory you're running your app and it can grow fast, so use it only for debugging purposes.

If you can get the doubling to happen again with the verbose mode active, please let me know.

Yoda49, 2016-11-08:
Ok. I try this.

Yoda49, 2016-11-08:
start sensor read (type=22, pin=23).
init peak: 137
update peak: 7 (138)
update peak: 8 (394)
update peak: 32 (398)
j=65, peak=398:
0 (137) 0 (137) 0 (137) 0 (137) 0 (137) 0 (137) 0 (138) 1 (394) 
0 (138) 0 (137) 1 (394) 1 (394) 0 (138) 1 (394) 1 (393) 1 (394) 
0 (133) 0 (137) 0 (137) 0 (138) 0 (137) 0 (138) 0 (137) 1 (393) 
0 (133) 0 (137) 0 (138) 0 (133) 0 (137) 0 (138) 1 (393) 1 (398) 
0 (136) 0 (137) 1 (393) 1 (394) 1 (393) 1 (395) 0 (137) 0 (137) 
0 (137) 0 (133) 0 (137) 0 (138) 0 (134) 0 (137) 0 (137) 0 (133) 
0 (138) 0 (137) 0 (133) 0 (137) 0 (138) 0 (137) 0 (138) 0 (137) 
0 (137) 0 (137) 0 (137) 0 (138) 0 (132) 0 (137) 0 (137) 0 (134) 

=> 1 37 1 3 (3c/3c) : OK
sensor type = 22, **temperature = 25.9**, humidity = 31.1
*** obtained readout successfully.
start sensor read (type=22, pin=23).
init peak: 137
update peak: 2 (138)
update peak: 7 (398)
j=64, peak=398:
0 (137) 0 (138) 0 (137) 0 (137) 0 (133) 0 (137) 1 (398) 0 (137) 
0 (133) 1 (391) 1 (395) 1 (394) 1 (394) 0 (133) 0 (137) 0 (138) 
0 (137) 0 (138) 0 (137) 0 (137) 0 (137) 0 (137) 1 (397) 0 (137) 
0 (133) 0 (137) 0 (137) 0 (134) 1 (393) 0 (138) 0 (138) 0 (137) 
1 (394) 0 (133) 0 (137) 0 (137) 0 (133) 1 (312) 0 (137) 0 (137) 
0 (131) 0 (138) 0 (137) 0 (137) 0 (137) 0 (137) 0 (137) 0 (137) 
0 (138) 0 (138) 0 (133) 0 (138) 0 (137) 0 (135) 0 (138) 0 (137) 
0 (132) 0 (137) 0 (138) 0 (133) 0 (138) 0 (137) 0 (137) 
=> 2 78 2 8 (84/84) : OK
sensor type = 22, **temperature = 52.0,** humidity = 63.2
*** obtained readout successfully.

Yoda49, 2016-11-08:
Another iteration:

start sensor read (type=22, pin=23).
init peak: 154
update peak: 7 (445)
update peak: 38 (446)
j=63, peak=446:
0 (154) 0 (146) 0 (153) 0 (154) 0 (153) 0 (154) 1 (445) 0 (153) 
1 (445) 0 (154) 0 (149) 0 (154) 0 (154) 0 (149) 0 (154) 0 (154) 
0 (154) 0 (153) 0 (154) 0 (153) 0 (153) 0 (154) 1 (406) 0 (154) 
0 (149) 0 (154) 0 (153) 0 (149) 0 (153) 0 (154) 0 (149) 0 (154) 
1 (445) 0 (153) 0 (154) 0 (150) 0 (153) 1 (446) 0 (154) 0 (149) 
0 (079) 0 (144) 0 (152) 0 (151) 0 (152) 0 (152) 0 (152) 0 (152) 
0 (152) 0 (152) 0 (152) 0 (148) 0 (152) 0 (152) 0 (148) 0 (152) 
0 (152) 0 (147) 0 (152) 0 (152) 0 (147) 0 (152) 
=> 2 80 2 0 (84/84) : OK
sensor type = 22, **temperature = 51.2, humidity = 64.0**
*** obtained readout successfully.
start sensor read (type=22, pin=23).
init peak: 138
update peak: 8 (398)
j=64, peak=398:
0 (138) 0 (137) 0 (137) 0 (138) 0 (138) 0 (133) 0 (138) 1 (398) 
0 (138) 0 (133) 1 (395) 1 (395) 1 (395) 0 (133) 0 (138) 1 (398) 
0 (138) 0 (002) 0 (137) 0 (137) 0 (138) 0 (137) 0 (137) 1 (398) 
0 (030) 0 (137) 0 (137) 0 (134) 0 (137) 0 (137) 1 (393) 0 (137) 
0 (137) 1 (394) 1 (394) 1 (394) 1 (393) 0 (138) 0 (138) 0 (138) 
0 (137) 0 (138) 0 (138) 0 (134) 0 (137) 0 (138) 0 (132) 0 (137) 
0 (137) 0 (133) 0 (137) 0 (138) 0 (133) 0 (137) 0 (137) 0 (137) 
0 (137) 0 (138) 0 (138) 0 (138) 0 (137) 0 (137) 0 (133) 
=> 1 39 1 2 (78/3d) : ERR
*** unexpected data: bits=64: 120 != 1 + 57 + 1 + 2

Yoda49, 2016-11-08:
full log: http://rgho.st/8W5gK5NKv

momenso, 2016-11-09:
Thanks @Yoda49! This is very helpful. I'll check and update this.

momenso, 2016-11-13:
Your logs shows exactly the bit shift problem. In these cases the CRC is affected as well as the data so it passes checking.

For example, where the data received corresponds to temperature = 52°C and humidity = 63.2% the CRC is correctly set to 132 which matches with the actual bits received.

[image]

To mitigate this problem in ```node-dht-sensor@0.0.32``` there's now a new constraint on the number of bits received to avoid accepting data with lost bits from the sensor that can cause this bit shifting to happen.

Yoda49, 2016-11-22:
It seems, everything is good!

momenso, 2016-11-22:
That's great! Thanks for the update.

closed 2016-11-22 by momenso