momenso/node-dht-sensor #54: Always returning 0, 0 for temperature and humidity

Soben, 2016-10-21:
I must be going crazy, but I can swear I've quadruple checked my breadboard configuration connectivity to my GPIO, but the results is always an array of temperature and humidity with the values of 0.

Any thoughts on this?

momenso, 2016-10-22:
There are a few GPIO numbering schemes used out there so this could be the problem. Because we are using the BCM2835 library we use its pinout numbering scheme which looks like this:
[image]
Can you share the details on which physical pin did you connect and what values are you specifying when you try to read the sensor? Are you using a DHT11 or DHT22?

Soben, 2016-10-25:
Hey momenso.

I will get this to you soon. I am still interested in resolving my problem, I just have been super occupied this weekend and so far this week.

I am using the DHT11. I will include a picture of my configuration, and include the script I used to call the sensor when I respond tonight.

Thanks,
-Chris.

Soben, 2016-10-27:
This is part of a bigger picture in which I'm trying to connect to my Nest too, and combine data in a useful way... but that's besides the point. My call is currently, simply:

`var currentDHTdata = dht.read(11, 4); // dhtSensor ID, GPIO ID
console.log(currentDHTdata.humidity);
console.log(currentDHTdata.temperature);`

My current set up (attached)... I realized last night that perhaps I am using the wrong side of the GPIO, but I can swear I set that up properly...

[image]

momenso, 2016-10-28:
Hi @Soben, according to the DHT11 datasheet you should be using a 4K7 Ω resistor and power it up from 3.5 up to 5.5 V. I see you have a 10k Ω resistor and it is connected to the 3.3V rail. So, perhaps you should try to adjust it accordingly to match the datasheet. Other than that all looks alright.

closed 2017-01-30 by momenso