momenso/node-dht-sensor #57: Not running at RPi3

abataille, 2016-11-13:
I have install BCM2835 successfully.
Then I install the node-dht-sensor.
From your test I ran
sudo node duty.js 22 17
BCM2835 initialized.
Error: failed to read sensor

The same works fine with an Raspberry Pi 1.
Any idea ?

momenso, 2016-11-13:
Hi @abataille. 
It is most likely this is a wiring issue, other than that, I see you enabled verbose mode so you should have more info on the log file (i.e. ```dht-sensor.log```).
If you followed the wiring diagram given as examples, please make sure you connected the sensor ground pin to GPIO Ground (e.g. pin 20, or any other ground available) and not GPIO 21.
[image]

abataille, 2016-11-13:
Data s Pin 17 and ground is the rightmost ground.
Is it possible that the bcm2835 library does not work correctly because the pi3 has a bcm2837 chip?

momenso, 2016-11-13:
Not at all. If you are using the latest bcm2835 library that should not be a problem.
What do you see on the ```dht-sensor.log``` file?

abataille, 2016-11-13:
I can check that tomorrow when I have access to that log.

abataille, 2016-11-14:
the log shows:
start sensor read (type=22, pin=17).
*** timeout #1

abataille, 2016-11-14:
I also changed the bcm2385 lib to v. 50 and the pin to pin 16.
The log now shows:
start sensor read (type=22, pin=16).
*** timeout #2

momenso, 2016-11-15:
It does look like the sensor is not connected (or not responding) on the pin you specified.
I recommend you double check your wiring. For example, if you're connecting to GPIO Pin 17 it should look like this:
[image]

abataille, 2016-11-15:
I already have this wiring, except I get 3.3V and ground from other pins.
Is there a chance that the bcm library does not work with the 64bit architecture the way we expect?

abataille, 2016-11-15:
Just to make sure that nothing is wrong with Jessie and the hardware.
I used a small phyton script i.e.
sudo ./AdafruitDHT.py 22 17
Temp=23.1*  Humidity=47.3%
That showed it somehow works.

momenso, 2016-11-15:
There should be no problem with BCM library on 64bit. I have the same setup here working fine.
```bash
pi@raspberrypi:~/node-dht-sensor $ uname -a
Linux raspberrypi 4.4.26-v7+ #915 SMP Thu Oct 20 17:08:44 BST 2016 armv7l GNU/Linux
pi@raspberrypi:~/node-dht-sensor $ lsb_release -d
Description:	Raspbian GNU/Linux 8.0 (jessie)
```
[image]
Simple read test:
```bash
pi@raspberrypi:~/node-dht-sensor $ node test/async-explicit.js 22 17 1
temperature: 22.4°C, humidity: 82.4%, time: 25ms
```
Also, if this python script from Adafruit is working it validates your wiring.

Can you try ```npm list``` to double check node-dht-sensor version?
You should get something like this:
```bash
pi@raspberrypi:~/node-dht-sensor $ npm list
node-dht-sensor@0.0.32 /home/pi/node-dht-sensor
└── nan@2.4.0
```

abataille, 2016-11-15:
My installation just looks the same.
Is there a chance that the modules are in too different locations?
node-dht-sensor ist located in /usr/lib/node_modules/node-dht-sensor
libbcm2835a is located in /usr/local/lib

momenso, 2016-11-15:
Your *libbcm2835.a* is in the default location and it seems that you installed node-dht-sensor globally, right? Anyway, that should not be a problem either. Although not common, you could try setting NODE_PATH to your node modules directory to make sure it is loading modules from where you expect, such as:
```bash
$ export NODE_PATH=/usr/lib/node_modules
```
One more thing we could check is node and npm versions you are running.
```bash
$ node -v
v7.1.0
$ npm -v
3.10.9
```

If all that is fine, here is a complete valitation procedure you can follow assuming BCM2835 library is already installed:
1) Download node-dht-sensor from github
```bash
pi@raspberrypi:~/project $ git clone https://github.com/momenso/node-dht-sensor.git
Cloning into 'node-dht-sensor'...
remote: Counting objects: 409, done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 409 (delta 6), reused 0 (delta 0), pack-reused 394
Receiving objects: 100% (409/409), 82.10 KiB | 0 bytes/s, done.
Resolving deltas: 100% (236/236), done.
Checking connectivity... done.
```
2) Go to the project folder and install the module
```bash
pi@raspberrypi:~/project $ cd node-dht-sensor/
pi@raspberrypi:~/project/node-dht-sensor $ npm install

> node-dht-sensor@0.0.32 preinstall /home/pi/project/node-dht-sensor
> ./check-lib.sh

Library bcm2835 found.

> node-dht-sensor@0.0.32 install /home/pi/project/node-dht-sensor
> node-gyp configure

> node-dht-sensor@0.0.32 postinstall /home/pi/project/node-dht-sensor
> node-gyp build

make: Entering directory '/home/pi/project/node-dht-sensor/build'
  CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o
  CXX(target) Release/obj.target/node_dht_sensor/dht-sensor.o
  SOLINK_MODULE(target) Release/obj.target/node_dht_sensor.node
  COPY Release/node_dht_sensor.node
make: Leaving directory '/home/pi/project/node-dht-sensor/build'
node-dht-sensor@0.0.32 /home/pi/project/node-dht-sensor
└── nan@2.4.0 
```
3) Execute a simple reading test (sensor DHT22 on GPIO pin 17).
```bash
pi@raspberrypi:~/project/node-dht-sensor $ node test/async-explicit.js 22 17 1
temperature: 22.7°C, humidity: 80.2%, time: 20ms
```

abataille, 2016-11-15:
Miracle. That worked.
Thank you for all your help !!

closed 2016-11-15 by abataille

JonHeron, 2019-02-23:
Hello,
`uname -a
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
lsb_release -d
Description:	Raspbian GNU/Linux 9.8 (stretch)
`
`npm list
(node:1467) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
node-dht-sensor@0.0.34 /home/pi/Downloads/node-dht-sensor
└── nan@2.8.0
node -v
v8.11.1
npm -v
1.4.21
(node:1482) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
`
[image]

I have been running a pi out in the greenhouse since last spring and I just noticed the temp/humidity DHT11 sensor stopped working.
I just figured the sensor went bad so I ordered another and... woops the sensor is fine...
I have a spare Pi so I decided to try to get a spare DHT11 running on a fresh install.
Still no joy. Something must have borked in an update, maybe? 
I tried your complete validation method above, it installs fine but it still cant read the sensor.
`pi@raspberrypi:~/Downloads/node-dht-sensor $ node test/async-explicit.js 11 04 1 
Error: failed to read sensor
`
dht-sensor.log is not present on my system.
Are there any other troubleshooting suggestions? 
Cheers,
Jon

JonHeron, 2019-02-24:
Solved, I had the wrong pinout for the 3 pin sensor. DOH!
Cheers,
Jon