momenso/node-dht-sensor #58: check-lib.sh fails on OSMC build 10.2016

mwittig, 2016-11-21:
Some users reported the same issue occurrs on Raspbian Jessie.

```
gcc -lbcm2835
/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crt1.o: In function `_start':
(.text+0x34): undefined reference to `main'
collect2: error: ld returned 1 exit status
```
Looks like an build issue with library `BCM2835`. Possibly related to <http://stackoverflow.com/questions/11116399/crt1-o-in-function-start-undefined-reference-to-main-in-linux>. It might be possible to work around issue by improving the error handling of the `check-lib.sh` script.

momenso, 2016-11-21:
Hi @mwittig. If you want to give it a shot feel free to submit a pull request. Contributions are welcome.

mwittig, 2016-12-05:
@momenso Sorry for the delay it reply. I have looked into this again and found the script works as expected. The actual problem was an build issue with the bcm2835 lib on OSMC. So, I am closing this now.

closed 2016-12-05 by mwittig