momenso/node-dht-sensor #61: problem with bcm2835_init

floxskit, 2017-01-24:
Hello !
i have a problem with my DHT22 sensor
After 5 minutes of working, i have this problem : 
bcm2835_init: gpio mmap failed: cannot allocate memory Failed to initialize sensor

I don't understand where is the problem.

Thank you !

momenso, 2017-01-24:
By any chance, are you creating a new instance of the sensor every time you read it?
It looks similar to #6. Can you confirm?

floxskit, 2017-01-24:
Where is this file ?

floxskit, 2017-01-29:
?

momenso, 2017-01-30:
Not sure what file you are referring to but you should run the [sample](https://github.com/momenso/node-dht-sensor#first-example) and see if it works for you. It is a good starting point.

floxskit, 2017-02-10:
I use Homebridge with your module node-dht-sensor and i don't know where are the file who take the temperature to change with your "sample"

closed 2017-12-03 by momenso