momenso/node-dht-sensor #62: Experiencing node-gyp build Problem

tommytwoeyes, 2017-02-08:
I have experienced an issue similar to previous users, during the `node-gyp build` process.

However, I do have the bcm2835 library installed properly.

During the node-dht-sensor installation process, the build script output this line:

`Library bcm2835 found.`

Here is the build script output:

`
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 4.4.38-v7+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build"
gyp ERR! cwd /home/pi/node_modules/node-dht-sensor
gyp ERR! node -v v0.12.6
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok
npm ERR! Linux 4.4.38-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "node-dht-sensor"
npm ERR! node v0.12.6
npm ERR! npm  v2.11.2
npm ERR! code ELIFECYCLE

npm ERR! node-dht-sensor@0.0.32 postinstall: `node-gyp build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-dht-sensor@0.0.32 postinstall script 'node-gyp build'.
npm ERR! This is most likely a problem with the node-dht-sensor package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp build
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-dht-sensor
npm ERR! There is likely additional logging output above.
`

momenso, 2017-02-08:
Are you perhaps trying to install as the root user? If so, try not doing that and using an unprivileged user.
Also, what Raspberry Pi model and OS you are running on?

andreicek, 2017-02-11:
I've fixed this issue for me by using a newer node (latest e.g.).

You can install it using the N tool (https://github.com/tj/n).

closed 2017-11-11 by momenso