momenso/node-dht-sensor #63: Error when reading sensor

AuspeXeu, 2017-04-24:
I receive this error message

> Error: failed to read sensor

When trying to read from my DHT22. However after several reboots of my raspberry pi it works again. Any way to debug this behaviour?

ThomasBormans, 2017-06-21:
I am having the same issue. A reboot fixes the problem most of the time. If this does not help, it is "magically" fixed after removing/readding the DHT22 physically from the breadboard.

AuspeXeu, 2017-06-21:
Yeah ... that's not really an option when you're operating a PI remotely :D

ThomasBormans, 2017-06-21:
I am also operating my Pi remotely so I feel your pain ;)

momenso, 2017-06-21:
How long and how often this happen?
What model of Pi and OS are you using?

tryshchenko, 2017-07-12:
I have the same error, but reconnect doesn't fix it.
I installed lib with --dht_verbose, but it doesn't print anything useful.
'BCM2835 initialized' and that's it'.
Afterward only 'Error: failed to read sensor'.
Is there any way to get the more detailed message?

AuspeXeu, 2017-07-12:
@momenso the first pi and raspbian as OS

ThomasBormans, 2017-07-12:
@momenso 
OS: Linux raspberrypi 4.4.38-v7+
Pi version: Pi 3 Model B

momenso, 2017-07-16:
I have been collecting readouts over the past three days without coming across this issue. I am planning to keep running for a few more days and also expand it from the first Raspberry Pi to include a Raspberry Pi 3 to see if I can reproduce this issue.

How often are you guys experiencing this?

ThomasBormans, 2017-07-17:
There is no pattern. On June 19th I encountered the problem three times. Two days later the problem occurred twice. Since then everything is going fine.

AuspeXeu, 2017-07-17:
I have switched to using wiringPI and a simple python script ... works without problems so far. Might be worth switching this module to wiringPI as well...

riatzukiza, 2017-11-02:
I have a dht11 that just does not seem to run, I get the same error, I've tried this library and node-rpio

arthurmoises, 2017-11-07:
In my case it logs this error "Error: failed to read sensor at Error (native)" sometimes. I'm using a resinOS 2.2.0+rev1 (prod) in a Raspberry PI Zero W. 

I've seen that earlier (issue #4 ), you mentioned about a "readout.isValid" method for checking if the device has read, but how do I implement it? 

Thanks in advance!

momenso, 2017-11-11:
If you use the sync read function then the isValid attribute is a property of the returned object.
For example, for a DHT11 connected to pin 4 you can execute the following code:
```javascript
var sensor = require('node-dht-sensor');

var readout = sensor.read(11, 4);
console.log(`temperature: ${readout.temperature.toFixed()}°C, ` +
              `humidity: ${readout.humidity.toFixed()}%, ` +
              `valid: ${readout.isValid}`);
```
This will output something like:
```bash
$ node app.js
temperature: 24°C, humidity: 47%, valid: true
```

AuspeXeu, 2017-11-12:
Does this also work for the async version?

momenso, 2017-11-12:
The async read function handles invalid readouts internally. That is, it retries to fetch sensor data up to three times before giving up. If after three read attempts it still fails then an error returned on the callback.
So, instead of the isValid attribute to check on, err will be set on the callback call.
For example, when trying to read the invalid port (i.e 7 instead of 4) with the following code.
```javascript
var sensor = require('node-dht-sensor');
sensor.read(11, 7, function(err, temperature, humidity) {
  if (err) console.error('' + err);
  else console.log("temperature: %s°C, humidity: %s%%",
      temperature.toFixed(), humidity.toFixed());
});
```
The output will be the following.
```
$ node app.js
Error: failed to read sensor
```

toddysm, 2017-12-03:
I am encountering the same issue as @momenso. The sensor, the wiring, and the board are all working if I use Python to read the data. I am using DHT11. It is unfortunate that this issue is not looked at since April :( Is this library still supported?

closed 2017-12-03 by momenso

toddysm, 2017-12-04:
@momenso Why was my comment deleted? I am still facing the issue

momenso, 2017-12-04:
Sorry @toddysm. It was my mistake.
Just open a new issue with your specifics (i.e. raspberry pi model, raspbian, node.js versions, logs and any error messages) and I will try to reproduce it.