momenso/node-dht-sensor #65: Not working in RPi3

mmontes11, 2017-10-21:
*** timeout #1
start sensor read (type=11, pin=17).
init peak: 0
bitCount=100, peak=0:
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000) 0 (000)
0 (000) 0 (000) 0 (000)
=> 0 0 0 0 (0/0) : OK
*** unexpected data: bits=100: 0 != 0 + 0 + 0 + 0
start sensor read (type=11, pin=17).
init peak: 0
bitCount=100, peak=0:

run-towards, 2017-11-01:
What sensor are you using? What pin convention following? BCM or WPi? From latter it's possible that you are reading from incorrect pin

momenso, 2017-11-11:
most likely incorrect pin configuration.

mmontes11, 2017-11-30:
I'm sure I have correct pin configuration but sometimes I got a timeout, I think the problem is in sensors

BTW how can I configure a WPi approach ?

saschakiefer, 2017-12-01:
I'm facing the exact same issue on the same hardware. The Python access is working for me as well. It's even working when I use the rpio library directly accessing the sensor with an own algorithm (since it's error prone I wanted to switch to the lib). 

#57 didn't fix my issue either.

momenso, 2017-12-03:
Currently, there's no support for other GPIO library other than the standard BCM2835 library. If someone wants to give that a try feel free to contribute.

battistaar, 2017-12-05:
Same for me, wiring configuration is correct (tested with AdafruitDHT).
[#57](https://github.com/momenso/node-dht-sensor/issues/57) didn't fix.

any news about it?

mmontes11, 2017-12-20:
#57 didn't fix my issue either.

any news?

fbrinker, 2018-01-11:
For me it is working on a RPI3 with BMC2835 and a DHT22
I am using this version: http://www.airspayce.com/mikem/bcm2835/bcm2835-1.52.tar.gz
Wired the DHT22 to Pin 11, GPIO 17 (so I pass 17 to the lib as second parameter)

YannLeBihan, 2018-01-15:
Hi,

It's also working here on a RPI3 + DHT11 with BMC2835, wired to Pin 7 - GPIO 4.

One particularity of the sensor I'm using (a Velleman VMA311) is its pins order : signal on the left, + in the middle and - on the right. Will soon test it with a DHT22, too.

And btw, @momenso, thanks a lot for this very useful module.

prasanpro, 2018-01-20:
It is working with RPi3 following instruction of installing node-gyp.

momenso, 2018-01-20:
If you still have any problems please open a new issue with your specifics.

closed 2018-01-20 by momenso