momenso/node-dht-sensor #66: Can not compile against bcm2835 objects: relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC

joergsesterhenn, 2017-11-22:
I tried installing node-dht-sensor 0.0.33 into this baseimage (https://github.com/resin-io-library/base-images/blob/c015aeeac8d0571dea9d9493ea229d565c306898/node/raspberry-pi2/debian/8.0/Dockerfile) and always run into this:

[Build]    > node-dht-sensor@0.0.33 preinstall /usr/src/app/node_modules/node-dht-sensor
[Build]    > ./check-lib.sh
[Build]    Library bcm2835 found.
[Build]    > node-dht-sensor@0.0.33 install /usr/src/app/node_modules/node-dht-sensor
[Build]    > node-gyp configure
[Build]    > node-dht-sensor@0.0.33 postinstall /usr/src/app/node_modules/node-dht-sensor
[Build]    > node-gyp build
[Build]    make: Entering directory '/usr/src/app/node_modules/node-dht-sensor/build'
[Build]      CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o
[Build]      CXX(target) Release/obj.target/node_dht_sensor/dht-sensor.o
[Build]      SOLINK_MODULE(target) Release/obj.target/node_dht_sensor.node
[Build]    /usr/bin/ld: //usr/local/lib/libbcm2835.a(bcm2835.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
[Build]    //usr/local/lib/libbcm2835.a: error adding symbols: Bad value
[Build]    collect2: error: ld returned 1 exit status
[Build]    node_dht_sensor.target.mk:123: recipe for target 'Release/obj.target/node_dht_sensor.node' failed
[Build]    make: Leaving directory '/usr/src/app/node_modules/node-dht-sensor/build'
[Build]    make: *** [Release/obj.target/node_dht_sensor.node] Error 1
[Build]    gyp ERR! build error
[Build]    gyp ERR! stack Error: `make` failed with exit code: 2
[Build]    gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
[Build]    gyp ERR! stack     at emitTwo (events.js:125:13)
[Build]    gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
[Build]    gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
[Build]    gyp ERR! System Linux 4.12.13
[Build]    gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build"
[Build]    gyp ERR! cwd /usr/src/app/node_modules/node-dht-sensor
[Build]    gyp ERR! node -v v8.8.1
[Build]    gyp ERR! node-gyp -v v3.6.2
[Build]    gyp ERR! not ok
[Build]    npm ERR! code ELIFECYCLE
[Build]    npm ERR! errno 1
[Build]    npm ERR! node-dht-sensor@0.0.33 postinstall: `node-gyp build`
[Build]    npm ERR! Exit status 1
[Build]    npm ERR!
[Build]    npm ERR! Failed at the node-dht-sensor@0.0.33 postinstall script.
[Build]    npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

I tried with multiple Versions of bcm2835 (1.52, 1.38) but it does not seem to make any difference.
I found this (https://stackoverflow.com/questions/13812185/how-to-recompile-with-fpic) stating it was an issue of mixing static and dynamically linked objects and tried  to export CXXFLAGS="$CXXFLAGS -fPIC", without success.

Please point me in the right direction. 
TIA Jörg

momenso, 2017-11-23:
I haven't tested resin yet, I don't expect it to be a problem though. In your docker file you are using ``armv7hf`` for a pi2, can you try ``armv7l`` instead?

joergsesterhenn, 2017-11-23:
Thanks for the fast reply!
I could not find any baseimages with armv7l for resin.

I just realize that I have a raspberry pi 2 Model B 1.1 which is **[BCM2836](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2836/README.md)**.

Is this supposed to work with node-dht-sensor?

momenso, 2017-11-26:
The BCM library is the same, the software supports different versions of the BCM chip.

I tried the same docker base image you used (``resin/raspberry-pi2-buildpack-deps:jessie``) and got the same problem when compiling but then I tried a different base image (``resin/rpi-raspbian``) and it worked.

Here is a node/express.js example: https://github.com/momenso/node-dht-sensor-resin-example

joergsesterhenn, 2017-11-27:
Thanks consider this issue closed then.

closed 2017-12-03 by momenso