momenso/node-dht-sensor #69: Error after sudo npm install -g node-dht-sensor

tomforsyth90, 2018-02-21:
As far as I can tell i've done everything right, but for some reason i keep getting this error when running that line of code:

`pi@tombabypi:~/bcm2835-1.55` $ sudo npm install -g node-dht-sensor
npm WARN lifecycle node-dht-sensor@0.0.34~preinstall: cannot run in wd %s %s (wd=%s) node-dht-sensor@0.0.34 ./check-lib.sh /usr/lib/node_modules/.staging/node-dht-sensor-c5090858

> node-dht-sensor@0.0.34 install /usr/lib/node_modules/node-dht-sensor
> node-gyp configure

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.13.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/node-dht-sensor/.node-gyp"

I have to get this resolved for a project that requires this... thanks

momenso, 2018-02-21:
Do you really want to install it as global package?

I do not recommend you do this, but if you want there sure are ways of doing it.

For example:

```sh
$ sudo npm install --unsafe-perm -g node-dht-sensor
```
When using it as a global package, also make sure you have ```NODE_PATH``` environment variable set to your global node_modules directory.

tomforsyth90, 2018-02-22:
Hi thanks for getting back to me.
To be honest I have no idea whether it should be installed globally or not all I am trying to do is follow this to no avail:

http://www.instructables.com/id/Raspberry-Pi-Nodejs-Blynk-App-DHT11DHT22AM2302/

Hope this helps explain my issue and stuff... Thanks

momenso, 2018-02-22:
That tutorial must have been written way back when root privileges were required for GPIO access.

Today, you can just install node-dht-sensor package at your project directory:

```sh
$ npm install node-dht-sensor
```
or you can add it as a dependency in your `package.json` file.

Either way, it should work without root access.

tomforsyth90, 2018-02-22:
@momenso, cheers for the reply.. I've just tried that and it still hasn't worked... this is what i'm getting now: :(

```
pi@tombabypi:~ $ cd bcm2835-1.55
pi@tombabypi:~/bcm2835-1.55 $ npm install node-dht-sensor

> node-dht-sensor@0.0.34 preinstall /home/pi/node_modules/.staging/node-dht-sensor-cfd4dcc2
> ./check-lib.sh

Library bcm2835 found.

> node-dht-sensor@0.0.34 install /home/pi/node_modules/node-dht-sensor
> node-gyp configure

> node-dht-sensor@0.0.34 postinstall /home/pi/node_modules/node-dht-sensor
> node-gyp build

make: Entering directory '/home/pi/node_modules/node-dht-sensor/build'
  CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o
  CXX(target) Release/obj.target/node_dht_sensor/dht-sensor.o
  SOLINK_MODULE(target) Release/obj.target/node_dht_sensor.node
  COPY Release/node_dht_sensor.node
make: Leaving directory '/home/pi/node_modules/node-dht-sensor/build'
/home/pi
└── node-dht-sensor@0.0.34 

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
```

momenso, 2018-02-22:
No errors, the installation looks fine.

However, you probably want to install the packages to your project's directory where you'll create your javascript project files.

Now that you already have BCM2835 library installed, you can for example:
```sh
$ mkdir ~/my-project
$ cd ~/my-project
$ npm install blynk-library
$ npm install node-dht-sensor
$ nano blynk-sensor-test.js
```
Here you should enter your javascript app code, making sure to update with your blynk auth token, sensor type and GPIO pin.

```javascript
var blynkLib = require('blynk-library');
var sensor = require('node-dht-sensor');

var AUTH = 'YOUR_AUTH_TOKEN';

// Setup Blynk
var blynk = new blynkLib.Blynk(AUTH);

// Sensor settings
var type = 11; // 11 for DHT11, 22 for DHT22 and AM2302
var pin  = 4;  // The GPIO pin number for sensor signal

// Start reading the sensor in 2 seconds intervals
setInterval(function() {
sensor.read(type, pin, function(err, temperature, humidity) {
    if (!err) {
        console.log('temperature: ' + temperature.toFixed(1) + 'C, ' +
            'humidity: ' + humidity.toFixed(1) + '%'
        );
	blynk.virtualWrite(3, temperature.toFixed(1));
	blynk.virtualWrite(4, humidity.toFixed(1));
    } else {
	console.warn('' + err);
    }
});
}, 2000);
```

Then just run it
```sh
$ node blynk-sensor-test.js
```
With that you should be good on the RPi side of things, of course you need to install blynk to your phone and configure it from there as well.

tomforsyth90, 2018-02-23:
Ahh that's brilliant! I've just done that and it has gotten further than anything i'd managed to do previously.. already got everything set up on my phone so just need to get this working and it should be spot on.

Just run it and i got this error:
pi@tombabypi:~/my-project $ node blynk-sensor-test.js
Connecting to: blynk-cloud.com 8441
SSL authorization...
Connected
Authorized
bcm2835_init: Unable to open /dev/mem: Permission denied
Segmentation fault

Do i just need to change the folder permissions?

momenso, 2018-02-23:
Are you using Raspbian? If so, it looks like it's old version. After Jessie released in Feb 2016 it should work without sudo. So, if you want to upgrade your distro that's another option.
Either way, it's great that it's working now!

tomforsyth90, 2018-02-23:
Yeah it's the latest version of Stretch...

closed 2018-02-24 by momenso