momenso/node-dht-sensor #71: dht on galileo

cristina21r, 2018-03-13:
Can I use this library to get temperature with Galileo gen 2??

Here's my code for  Pi3:

var opcua = require("node-opcua");
var w1bus = require("node-w1bus");
var bus = w1bus.create();
var rpiDhtSensor = require("rpi-dht-sensor");

function read_sensor(sensor,callback) {
                bus.getValueFrom(sensor)
                .then(function(res){  callback(null,res.result); })
                .catch(callback);
        };

function start_server() {

        var server = new opcua.OPCUAServer({
                port: 1234
        });

        server.buildInfo.productName ="Rapsberry OPCUA Server"

        function post_initialize() {

                var addressSpace = server.engine.addressSpace;
                var devices = addressSpace.addFolder("ObjectsFolder",{ browseName: "Devices"});
                var temperatureSensor = "28-041460c2a3ff";
                install_sensor(temperatureSensor);
        };

        var tempValue = -20.0;
        var tempValue1 = -20.0;
        var humValue = -20.0;

        function install_sensor(sensor) {
                var addressSpace = server.engine.addressSpace;
                var sensorName = "ds18b20";
                var sensorName1 = "dht11";
                var devices = addressSpace.rootFolder.objects.devices;
                var sensor  = addressSpace.addObject({
                        organizedBy: devices,
                        browseName:  sensorName
                });

                var dht = new rpiDhtSensor.DHT11(27);

                setInterval(function() {
                        var temperatureSensor = "28-041460c2a3ff";
                        read_sensor(temperatureSensor,function(err,result) {
                                tempValue = result.value;
                        });
                        var readout = dht.read();
                        tempValue1 = readout.temperature.toFixed(2);
                        humValue = readout.humidity.toFixed(2);
                },3000);

                addressSpace.addVariable({
                        propertyOf: sensor,
                        browseName: "temperature1",
                        dataType:   "Double",
                        value: {
                                get: function () {
                                        return new opcua.Variant({dataType: opcua.DataType.Double, value: tempValue });
                                }
                        }
                });

                addressSpace.addVariable({
                        propertyOf: sensor,
                        browseName: "temperature2",
                        dataType:   "Double",
                        value: {
                                get: function () {
                                        return new opcua.Variant({dataType: opcua.DataType.Double, value: tempValue1 });
                                }
                        }
                });

                addressSpace.addVariable({
                        propertyOf: sensor,
                        browseName: "humidity",
                        dataType: "Double",
                        value: {
                                get: function() {
                                        return new opcua.Variant({dataType: opcua.DataType.Double, value: humValue });
                                }
                        }
                });
        };

        server.initialize(post_initialize);

        server.start(function() {
                console.log("\n\nTemperature - OPCUAServer");
                var endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
                console.log("Server endpoint url is ", endpointUrl );
                console.log("\n");
        });
}

start_server();

I want to make it work on galileo...some help?

momenso, 2018-03-13:
Sorry but node-dht-sensor uses the BCM2835 library for GPIO access and it is designed to work on Raspberry only.

cristina21r, 2018-03-13:
...and do you know other way to get those values from sensors on galileo..?

momenso, 2018-03-14:
I'm not aware of any readily available solutions. It will require some development work to be able to read and decode the sensor data.

momenso, 2019-04-28:
Let's open a feature request if there's demand for it.

closed 2019-04-28 by momenso