momenso/node-dht-sensor #72: Return Values

SethRzeszutek, 2018-04-18:
Am I viewing this wrong?
 I am trying to actually store the data, but nothing gets written to it.

So if I use the following now value overwrites the 0.
`function readingTemp()
{
	var sensor = require('node-dht-sensor');
	var temp = 0;
	sensor.read(11, 14, function(err, temperature, humidity) {
    		if (!err) {
        		//console.log('temp: ' + (temperature.toFixed(1)*9/5+32) + '°F, ' + 'humidity: ' + humidity.toFixed(1) + '%');
			temp = (temperature.toFixed(1)*9/5+32); 
    		}});
	return(temp);
}
var temp2 = 0;
temp2=readingTemp();
console.log(temp2);
//readingTemp();
`
OUTPUT:
0

However, if I assign it to a console.log, it does store it.
`
function readingTemp()
{
	var sensor = require('node-dht-sensor');
	var temp = 0;
	sensor.read(11, 14, function(err, temperature, humidity) {
    		if (!err) {
        		//console.log('temp: ' + (temperature.toFixed(1)*9/5+32) + '°F, ' + 'humidity: ' + humidity.toFixed(1) + '%');
			temp = (console.log(temperature.toFixed(1)*9/5+32)); 
    		}});
	return(temp);
}
var temp2 = 0;
temp2=readingTemp();
console.log(temp2);
//readingTemp();
`
OUTPUT:
0
71.6

Even for this last part, it still pushes out the 0.

jhoughjr, 2019-01-04:
It appears you are setting temp to the return value of console.log, which it doesnt return anything.

jhoughjr, 2019-01-04:
also you should pst code as ''' //some code here ''' makes it easier to read.

jhoughjr, 2019-01-04:
my bad ``` // some code ```. use the ` char, the tilde unshifted.

SethRzeszutek, 2019-01-11:
You're right, I figured that out after a long night of troubleshooting and never got back to updating that.

SethRzeszutek, 2019-01-18:
Resolved #72

closed 2019-01-18 by SethRzeszutek