momenso/node-dht-sensor #73: node-dht-sensor works but seems to crash other running apps every time it is invoked

davidhitchins, 2018-04-29:
Not sure why this is happening. I have set up a DHT22 temp humidity sensor according to the instructions (except I am used GPIO 22 then, when this caused problems, GPIO 23) on a Raspberry pi zero W.
I have three other apps running (automatically restarted by PM2 should they crash) and they run fine until I run the node js file temphumidmonitor.js to print the temp, humidity and time to the console (no other actions at present - just a simple script which is attached). Then they are restarted by PM2. 
I have also attached a screenshot of the Putty terminal showing how everything is ok till I run the script with dht-sensor. Even if the script runs fine and spits out the temp and humidity reading without error, the other apps still restart. However a good 30% of the time dht-sensor encounters an error which, according to the dht-sensor-log file (I installed it in verbose mode) is something like this:

0 (208) 0 (208) 0 (208) 0 (209) 0 (208) 0 (209) 1 (589) 0 (200) 
0 (208) 0 (209) 1 (589) 1 (589) 0 (208) 0 (208) 0 (208) 0 (200) 
0 (203) 0 (205) 0 (204) 0 (204) 0 (204) 0 (204) 0 (197) 0 (150) 
1 (588) 0 (208) 0 (208) 1 (588) 0 (208) 0 (208) 0 (200) 1 (589) 
1 (589) 1 (589) 1 (588) 1 (589) 0 (208) 1 (589) 0 (200) 
=> 2 30 0 91 (7a/c3) : ERR
*** unexpected data: bits=40: 122 != 2 + 48 + 0 + 145

It is NOT this error that makes the other apps restart; they do anyway even if the result is like this:

start sensor read (type=22, pin=23).
init peak: 208
update peak: 7 (588)
bitCount=41, peak=588:
0 (208) 0 (208) 0 (208) 0 (208) 0 (208) 0 (208) 1 (588) 0 (200) 
0 (208) 0 (208) 1 (588) 0 (208) 1 (588) 1 (588) 1 (588) 1 (581) 
0 (208) 0 (208) 0 (208) 0 (208) 0 (208) 0 (208) 0 (208) 0 (200) 
1 (588) 1 (588) 0 (208) 0 (209) 1 (588) 0 (208) 0 (208) 0 (200) 
1 (588) 1 (588) 1 (588) 1 (588) 1 (588) 0 (208) 0 (208) 1 (580) 

=> 2 2f 0 c8 (f9/f9) : OK
sensor type = 22, temperature = 20.0, humidity = 55.9
*** obtained readout successfully.
[image]

Here's my version of the script I am using:
`//const mysqlfunctions = require('./scripts/mysqlfunctions');
//var connection = require('./scripts/db');
var sensor = require('node-dht-sensor');
var moment = require('moment');

//var sql = "CALL `DefaultTempHumidityRecInterval`()";

//connection.query(sql, function(err, result){

  //if (err) throw err;

  //var interval = result[0][0].TEMPHUMINTERVAL;

//console.log(interval);

//});
//setInterval(recordData, 10000);

//Using GPIO 22 (PHYSICAL PIN 15)
//function recordData() {

    //console.log ("running");
     sensor.read(22, 23, function(err, temperature, humidity) {
       if (!err) {
        console.log('temp: ' + temperature.toFixed(1) + '°C, ' +
              'humidity: ' + humidity.toFixed(1) + '%' + ' at ' + moment().format('MMMM Do YYYY, h:mm:ss a')
          );

        //mysqlfunctions.recTemperatureEvent(1, 7, temperature.toFixed(1));
      //mysqlfunctions.recHumidityEvent(1, 7, humidity.toFixed(1));
     }
 });
//}

How can I find out why this script makes the other running apps restart? 
Merci Beaucoup!

davidhitchins, 2018-04-30:
I have answered my own question!
Because I am using PM2 with the --watch option and because I installed dht-sensor with the verbose logging option and because dht-sensor.log is in the same folder as my programme (often the case?) then every time a record is taken it updates the dht-sensor.log file which triggers PM2 to restart all the apps in that folder and subfolders. That is solved.
BUT there is still the issue of 
*** unexpected data: bits=40: 122 != 2 + 48 + 0 + 145
happening.
It does not seem to be very important but maybe it is?

momenso, 2018-05-01:
The unexpected data message is reported when the data checksum fails. If that doesn't happen often it should not be a problem. The causes for this may vary and include:
- malfunctioning sensor - there are many manufacturers in the market and overall quality is not the same;
- electrical interference between sensor and GPIO - sometimes caused by use of long wires;
- not enough processing power - decoding data from the sensor is not processing intensive but it is time sensitive and if the processor is too busy doing something else it may lose some data.

However, in most situations, these errors are handled internally and your app should be able to fetch a readout without problems.

momenso, 2019-05-01:
Closing this one after one year without updates.

closed 2019-05-01 by momenso