momenso/node-dht-sensor #74: Can't install package

filipecosta-dev, 2018-06-10:
Hello, I am trying to install this package to work with it for my Raspberry Pi, for now it seems there is an issue with it:
> npm install --save node-dht-sensor
>  ./check-lib.sh
>'.' is not recognized as an internal or external command,
operable program or batch file.
...
> npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-dht-sensor@0.0.34 preinstall: `./check-lib.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-dht-sensor@0.0.34 preinstall script.

Any clue to fix it?

momenso, 2018-06-10:
Try installing it on [Raspbian](https://www.raspbian.org/) - that should work.

filipecosta-dev, 2018-06-10:
Alright, that makes sense since I am using Windows at the moment, I will test it and comeback to you shortly.