momenso/node-dht-sensor #75: Incorrect temp and humidity values.

simvn, 2018-12-11:
Hi everybody,

I just set up my raspberry, installed the libraries and ran the the **first example**. 

However, I receive all the time **temp: -40.0°C and humitdity 1.0%** values. Humidity shows small changes when I am blowing at the dht22 sensor, but immediatley returns to 1.0%.

Here is my setup:

Raspberry 3b+
OS: Raspbian Stretch (4.1.4)
Sensor: DHT22
Node version: 10.14.1
bcm2835 version: 1.58

Any help is highly appreciated!

Thank you and best 
Simon

simvn, 2018-12-11:
I have tested it now with adafruitDHT as well and it shows the same values.
Thus, I assume something is wrong with my sensor :/

FifthRooter, 2018-12-17:
@simvn Close the issue then, especially if it ended up not being related to this package :) Good luck with your project!

simvn, 2018-12-21:
@FifthRooter thanks.

I received a new sensor today and now it's working. So the old sensor was trash.

closed 2018-12-21 by simvn