momenso/node-dht-sensor #77: Hardware emulation

kothique, 2019-04-27:
I'm writing a web server that uses your library. The problem is development is being done on my PC which has no GPIO's, of course, so I can't test it if I install the library. Do you think it is possible to emulate sensors if GPIO's are not present?

https://github.com/jperkin/node-rpio, for example, has a `mock` option:

```js
var options = {
        gpiomem: true,          /* Use /dev/gpiomem */
        mapping: 'physical',    /* Use the P1-P40 numbering scheme */
        mock: undefined,        /* Emulate specific hardware in mock mode */
}
```

I'm not really familiar with the 'node <-> C/C++'  stuff, so my workaround doesn't seem too clear and legit: https://github.com/daisy-electronics/dht-sensor/commit/f68c1f6acb6f89411da18039265d30d099fb9bdd

Maybe you could turn me in the right direction to implement this properly, so I could create a pull request in the future?

momenso, 2019-04-28:
It does seems like a legit need to mock out the sensor part of the app. I like the idea.

Perhaps we could try and move away from using the *BCM2835 library*. Using instead [node-rpio](https://github.com/jperkin/node-rpio) does look like a good candidate, what you think?

kothique, 2019-04-30:
> It does seems like a legit need to mock out the sensor part of the app. I like the idea.
> 
> Perhaps we could try and move away from using the _BCM2835 library_. Using instead [node-rpio](https://github.com/jperkin/node-rpio) does look like a good candidate, what you think?

Sounds like a good idea in general.

`node-rpio` actually has an example that should work with `DHT11`: https://github.com/jperkin/node-rpio/blob/master/examples/dht11.js Though it didn't work for me. None of the solutions here did either: https://github.com/jperkin/node-rpio/issues/57 Can you try it out?

kothique, 2019-04-30:
(not related to sensor reading itself)

Also, if we move to using `node-rpio`, it's gonna break compatibility as different pin numbering may be used depending on how `node-rpio` was inintialized. So in the new version we could probably rename synchronous `read` to `readSync`, and make `read` return a Promise if no callback is passed.

momenso, 2019-04-30:
I did a quick test and example from rpio for reading a DHT11 sensor didn't work for me either. That's weird. I'll try to debug it to see if I can find something.

As for the port numbering we may be able to initialize rpio to use tge GPIO mapping. I saw this example from the docs:
```javascript
rpio.init({mapping: 'gpio'});   /* Use the GPIOxx numbering */
```
So, this really works there should be no problem with breaking compatibility.

kothique, 2019-04-30:
I mean people would want to use `node-dht-sensor` along with `node-rpio`, so we cannot initialize `node-rpio` ourselves. Thus we cannot be sure what numbering is used.

kothique, 2019-04-30:
Reading DHT sensor data requires precise timing, so I'm worried that node's function call overhead may cause it to be less reliable. Am I wrong?

momenso, 2019-04-30:
You're right. It might be a problem but I wanted to test it first to see if it is really a problem.

momenso, 2019-05-01:
I've got some valid readouts with a modified `dht.js` from *rpio* examples! Definitely not nearly as reliable as using *node-dht-sensor* but it did work after tweaking the timing and bit level threshold.

I've changed the following:
  1. MCU initialization timings to **5 ms** and **9 ms**, for HIGH and LOW respectively;
  2. Bit length cutoff to **> 350**;
  3. Bit level 1 when length **> 150** and 0 otherwise.

Any luck on your side?

kothique, 2019-05-01:
No luck unfortunately. All I found out is if MCU intialization LOW timing is less than 11ms, then DHT response signal is always HIGH.

kothique, 2019-05-01:
Anyway, even if I make it work for myself, calibrating these values for every different device, load, and lunar phase is not an option, is it?

momenso, 2019-05-01:
Yes, it will be tricky to make it work reliably - I think we should park the rpio idea for now.

Lets go back to providing a mocking mode where we can provide fake readout values. Perhaps that is going to be useful, not a hardware emulation tough. The goal would be to enable it possible to develop an app when not having an actual **DHT** sensor connected to the **GPIO**, right?

kothique, 2019-05-01:
More like not having GPIO's whatsoever.

PR #86 (2019-05-02): Support for test mode
As suggested by @kothique, the goal is to provide a new operation mode for using the *node-dht-sensor* -  test mode. When enabled, the library will not communicate with sensor hardware but instead it will provide predefined readout values. This will enable development and testing without an actual sensor connected.

To enable the test mode, fake values must be defined at initialization. In the example below we specify fixed values for temperature equal to 21ºC and humidity equal to 60%.
```javascript
sensor.initialize({
    test : {
        fake: {
            temperature [... cut]
closed 2019-05-02 by momenso