momenso/node-dht-sensor #78: ./check-lib.sh

FelipeGerolomo, 2019-04-28:
pm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-dht-sensor@0.0.36 preinstall: `./check-lib.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-dht-sensor@0.0.36 preinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\{USERNAME}\AppData\Roaming\npm-cache\_logs\2019-04-28T02_18_51_374Z-debug.log

kothique, 2019-04-28:
You should post the log here (C:\Users{USERNAME}\AppData\Roaming\npm-cache_logs\2019-04-28T02_18_51_374Z-debug.log)

momenso, 2019-04-28:
Hi @FelipeGerolomo, It does look like you're trying to run it on a PC running windows.
Please try following these [instructions](https://github.com/momenso/node-dht-sensor/blob/master/README.md) on a raspberry pi device running raspbian OS.

kothique, 2019-05-10:
@FelipeGerolomo, any news since then?

momenso, 2020-06-07:
If this issue is still relevant, feel free to update it.

closed 2020-06-07 by momenso