momenso/node-dht-sensor #101: node-dht-sensor broken on RPI4

jgarvey955, 2019-07-08:
After updating to buster on my RPI3B+ my system was running fine. All other peripherals work fine except when i shift the 3B+ out for the 4B i get an 2835 MMAL error saying that memory can not be mapped. when i swap the 4B back to the 3B+ it runs fine. Is there something new???

momenso, 2019-07-08:
We had a similar problem reported in #16. 

Perhaps you can run a simple test code (e.g. the [First Example](https://github.com/momenso/node-dht-sensor#first-example) from the README file) to confirm if you still have the problem.

jgarvey955, 2019-07-09:
I have confirmed with Raspberry Pi Org that the bcm2835 library will need to be rewritten for the rpi4b due to changes with the chip.

https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=245107&p=1495602&sid=2abe572cd359aafd8155eec8705e1f5d#p1495602

momenso, 2019-07-09:
Thanks for the update @jgarvey955!

ronasick, 2019-08-17:
How would you use this library with the new bcm2835 1.60 version? Is an update of this library needed?

vladimirvolek, 2019-08-22:
I have the same problem. Is there any workaround?

[image]

schadokar, 2019-09-01:
I am experiencing the same issue. I am using DHT22 and raspberry pi 4.

momenso, 2019-09-01:
Due to the limited availability of the RPi 4 I have been unable to work on this issue. If any lucky owner of a RPi4 is interested in working on this issue it will be most welcome! If not, I am stuck waiting for my order to be fullfilled, hopefully soon.

ronasick, 2019-09-01:
Excited for the update! This is my current workaround
[image]

closed 2019-09-22 by momenso

momenso, 2019-09-22:
Fix for running **node-dht-sensor** on the Raspberry Pi 4 is now released. 
Please make sure you upgrade to version **0.4.0** and check if it works for you.

PR #104 (2019-09-22): Update bcm2835 lib to 1.60
Update BCM2835 library to version 1.60 that is compatible with the Raspberry Pi 4.

Fixes #101.