momenso/node-dht-sensor #107: DHT22 reading not working

felicyta22, 2020-04-22:
Hi,

I am using a raspberry pi4 with the DHT22 sensor. 
I am running the example program 
```
var sensor = require("node-dht-sensor");

sensor.read(22, 4, function(err, temperature, humidity) {
  if (!err) {
    console.log(`temp: ${temperature}°C, humidity: ${humidity}%`);
  }
});
```
I installed the node-dht-sensor with debug mode and I see only: 
`BCM2835 initialized.`
I tried switching the sensor.read(11,4) and then I get some reading (obviously wrong because there is nothing connected) But at least it reads something.
If I change the device to 22 then I only see the above log and then nothing.

I tried then using the python DHT ada fruit library and it worked (sudo pip3 install Adafruit_DHT) I got correct readings from the sensor. Any hints what can be wrong?
the DHT22 which I'm using has only three pins. I connected it to the GPIO 4, hardware pin 7 on the raspberry pi.

momenso, 2020-04-22:
Hi @felicyta22,

Here is an example of running the same operation with the *node-dht-sensor* and the *Adafruit_DHT* libraries using an *DHT22* sensor connected to *GPIO 4*.

1) Using *node-dht-sensor*:
```bash
pi@raspberry:~/node-dht-sensor $ node examples/async-explicit.js 22 4 1
temperature: 24.3°C, humidity: 51.1%, time: 27ms
```
2) Using *Adafruit_DHT*:
```bash
pi@raspberry:~/Adafruit_Python_DHT/examples $ sudo ./AdafruitDHT.py 22 4
Temp=25.0*  Humidity=50.2%
```
As you can see both work seamlessly. So, if have problem running the same examples above I encourage you to check the following:
- Confirm that you are using the correct GPIO number (not the physical pin number);
- Have latest version of the libraries.

felicyta22, 2020-04-23:
Thanks, the second option worked well for me. I don't know why the first one doesn't (using the example code I posted). Thanks again for your quick response! I think that will do for now, the sensor is working and nicely sending data.

closed 2020-04-23 by felicyta22

Sly1024, 2020-07-14:
Hi! 

I have a similar problem. I have a Raspberry Pi4 B with two DHT22 sensors, running raspbian:

```
Linux home 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
Raspbian GNU/Linux 10 (buster)
```

I installed node 10 (v10.21.0) and installed the latest node-dht-sensor library. Interestingly it always failed at the post-install node-gyp build step. I then did this:

```
npm install node-dht-sensor --ignore-scripts
cd node_modules/node-dht-sensor
node-gyp configure
node-gyp build
```

and it went fine (I have node-gyp installed  globally).

If I try any of the example scripts, I don't get any result. The promise example never resolves the promise. No error, just nothing.
With the loop example that reads the sensors every 2 seconds I get:

```
BCM2835 initialized.
[Indoor] temperature: 0.0°C, humidity: 0.0%
[Outdoor] temperature: 0.0°C, humidity: 0.0%
[Indoor] temperature: 0.0°C, humidity: 0.0%
[Outdoor] temperature: 0.0°C, humidity: 0.0%
```

I'm sure the pin numbers (15, 18) are OK, because:

```
root@home:/home/pi/adafruit/Adafruit_Python_DHT/examples# ./AdafruitDHT.py 22 15
Temp=26.5*  Humidity=46.3%
root@home:/home/pi/adafruit/Adafruit_Python_DHT/examples# ./AdafruitDHT.py 22 18
Temp=27.5*  Humidity=45.2%
```

Any ideas what I'm missing?