momenso/node-dht-sensor #108: Installation error

grucha532, 2020-04-25:
I got an error while trying to install the node for the DHT22 sensor. **Neither**
`pi@raspberrypi:~ $ npm install node-dht-sensor
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code EACCES
npm ERR! syscall unlink
npm ERR! path /home/pi/node_modules/node-dht-sensor/build/Makefile
npm ERR! errno -13
npm ERR! Error: EACCES: permission denied, unlink '/home/pi/node_modules/node-dht-sensor/build/Makefile'
npm ERR!  [OperationalError: EACCES: permission denied, unlink '/home/pi/node_modules/node-dht-sensor/build/Makefile'] {
npm ERR!   cause: [Error: EACCES: permission denied, unlink '/home/pi/node_modules/node-dht-sensor/build/Makefile'] {
npm ERR!     errno: -13,
npm ERR!     code: 'EACCES',
npm ERR!     syscall: 'unlink',
npm ERR!     path: '/home/pi/node_modules/node-dht-sensor/build/Makefile'
npm ERR!   },
npm ERR!   stack: "Error: EACCES: permission denied, unlink '/home/pi/node_modules/node-dht-sensor/build/Makefile'",
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'unlink',
npm ERR!   path: '/home/pi/node_modules/node-dht-sensor/build/Makefile'
npm ERR! }
npm ERR!
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR!
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-25T10_40_30_125Z-debug.log`

**nor**
`pi@raspberrypi:~ $ npm install node-dht-sensor --unsafe-perm
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code EACCES
npm ERR! syscall unlink
npm ERR! path /home/pi/node_modules/.node-dht-sensor.DELETE/build/Makefile
npm ERR! errno -13
npm ERR! Error: EACCES: permission denied, unlink '/home/pi/node_modules/.node-dht-sensor.DELETE/build/Makefile'
npm ERR!  [OperationalError: EACCES: permission denied, unlink '/home/pi/node_modules/.node-dht-sensor.DELETE/build/Makefile'] {
npm ERR!   cause: [Error: EACCES: permission denied, unlink '/home/pi/node_modules/.node-dht-sensor.DELETE/build/Makefile'] {
npm ERR!     errno: -13,
npm ERR!     code: 'EACCES',
npm ERR!     syscall: 'unlink',
npm ERR!     path: '/home/pi/node_modules/.node-dht-sensor.DELETE/build/Makefile'
npm ERR!   },
npm ERR!   stack: "Error: EACCES: permission denied, unlink '/home/pi/node_modules/.node-dht-sensor.DELETE/build/Makefile'",
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'unlink',
npm ERR!   path: '/home/pi/node_modules/.node-dht-sensor.DELETE/build/Makefile'
npm ERR! }
npm ERR!
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR!
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-25T10_42_33_146Z-debug.log`

work. Does anyone have any idea how to fix it?
Thanks in advance!

momenso, 2020-04-25:
It does look like you have a problem with your node/npm installation. Can you install any other npm packages or is it just with this particular one?

momenso, 2020-06-07:
I hope you sorted out this problem, if not feel free to send an update.

closed 2020-06-07 by momenso