momenso/node-dht-sensor #111: Modul displays valid data even when sensor is disconnected

phffmn, 2020-06-04:
I'm using a dht22 and if I use your multi sensor example let it run and disconnect one sensor while it's running it will output the last collected data indefinitely. It should throw an error or at least display information that no data could be retrieved.

momenso, 2020-06-07:
Hi @phffmn.

You are right. The current default behaviour when using the synchronous read API is to cache the last readout value but that can be validated very simply. The multi sensor example does not really handle any errors. 
One possible way to extend that example is by checking the `isValid` attribute of the readout value.
```javascript
var readout = sensorLib.read(
    this.sensors[sensor].type,
    this.sensors[sensor].pin
);
if (readout.isValid) {
    console.log(
        `[${this.sensors[sensor].name}] ` +
        `temperature: ${readout.temperature.toFixed(1)}°C, ` +
        `humidity: ${readout.humidity.toFixed(1)}%`
    );
} else {
    console.error(`Failed to read sensor name='${this.sensors[sensor].name}'`);
}
```
I would encourage you to add that kind of validation to your code if the cached value is not what you are looking for.

phffmn, 2020-06-08:
Thank you very much, that should fix the problem.

closed 2020-06-08 by phffmn