momenso/node-dht-sensor #116: Timeouts reading from sensor

brlnr23, 2020-10-26:
I have searched a lot in this repository but unfortunately I couldn't find a solution for the problem I am facing. 

I constantly get an error reading data from the sensor: ```Error: failed to read sensor```

First off
1) Using the Adafruit example script returns the values I am expecting - so it seems that its not a wiring issue
2) I am using a DHT22 sensor (as module) without a pull up resistor 
3) ```npm -v: 6.14.8``` / ```node -v: v9.11.1```
4) I have installed the package with the verbose flag ```npm install node-dht-sensor --dht_verbose=true```

If I run that simple script (taken from the examples) it returns an error. The logfile just says 
``` 
start sensor read (type=22, pin=14).
*** timeout #2
```

```
sensor = require("node-dht-sensor");
var usage =
  "USAGE: node async-explicit.js [sensorType] [gpioPin] <repeats>\n" +
  "    sensorType:\n" +
  "         11: For DHT11 sensor.\n" +
  "         22: For DHT22 or AM2302 sensors.\n\n" +
  "    gpipPin:\n" +
  "         Pin number where the sensor is physically connected to.\n\n" +
  "    repeats:\n" +
  "         How many times the read operation will be performed, default: 10\n";

if (process.argv.length < 4) {
  console.warn(usage);
  process.exit(1);
}

var sensorType = parseInt(process.argv[2], 10);
var gpioPin = parseInt(process.argv[3], 10);
var repeats = parseInt(process.argv[4] || "10", 10);
var count = 0;
var start = 0;
var end = 0;

var iid = setInterval(function() {
  if (++count >= repeats) {
    clearInterval(iid);
  }

  start = new Date().getTime();

  sensor.read(sensorType, gpioPin, function(err, temperature, humidity) {
    end = new Date().getTime();
    if (err) {
      console.warn("" + err);
    } else {
      console.log(
        "temperature: %s°C, humidity: %s%%, time: %dms",
        temperature.toFixed(1),
        humidity.toFixed(1),
        end - start
      );
    }
  });
}, 3000);
```

Results in:  
```
node index.js 22 14 1
(node:5316) Warning: N-API is an experimental feature and could change at any time.
BCM2835 initialized.
Error: failed to read sensor
```

If I run the same script but with type 11 it returns completely wrong values, but it returns at least something
```
node index.js 11 14 1
(node:5326) Warning: N-API is an experimental feature and could change at any time.
BCM2835 initialized.
temperature = 0, humidity = 2
temperature: 0.0°C, humidity: 2.0%, time: 52ms
```

To me this looks like an issue with the DHT22 type of sensor. Any hints are welcome. I am happy  to provide more information if needed

cainpsycode, 2020-12-23:
Hello! I also had the same error and found a solution for myself. I found a difference between the dht-sensor implementation and the protocol, in particular the low trigger value timeout it should be within 1-18ms. I have increased the low voltage latency from 800ms to 1000ms.
changes:
https://github.com/cainpsycode/node-dht-sensor/commit/50811e3ef593ed9b91a1fa2d17a1b4773bde549b

oscar-c, 2021-01-11:
> Hello! I also had the same error and found a solution for myself. I found a difference between the dht-sensor implementation and the protocol, in particular the low trigger value timeout it should be within 1-18ms. I have increased the low voltage latency from 800ms to 1000ms.
> changes:
> [cainpsycode@50811e3](https://github.com/cainpsycode/node-dht-sensor/commit/50811e3ef593ed9b91a1fa2d17a1b4773bde549b)

I too had this problem. The mentioned changes fixed my problem as well.

brlnr23, 2021-02-01:
thanks for this hint. Since I am no c++ guy at all: am I right that I need to complile the cpp files? And if so: do you have any hints how a bloody starter is able to do so ;)

cainpsycode, 2021-02-01:
> thanks for this hint. Since I am no c++ guy at all: am I right that I need to complile the cpp files? And if so: do you have any hints how a bloody starter is able to do so ;)
Yes you need rebuild native cpp code for your target platform. Just follow the instruction https://github.com/momenso/node-dht-sensor#reference-for-building-from-source

brlnr23, 2021-02-01:
a big up to you, @cainpsycode! After some back and forth it worked for me!

If someone else wants to build it from source:
I did that on my Raspberry directly. For me, there was to need to run the following command
```
$ sudo update-alternatives --install /usr/bin/node-gyp node-gyp /opt/node-v10.15.3-linux-armv7l/bin/node-gyp 1
```

Instead I need to run the following commands as sudo in the node-dht-sensor directory
```
$ sudo node-gyp configure
$ sudo node-gyp build
```

I was also missing the package `node-addon-api` but a simple `npm install node-addon-api` fixed it. 

Once the build process has been completed I could import the self-made build via - in my case with promises - 
```
var sensor = require("./lib").promises;

... do fancy things here
```

cainpsycode, 2021-02-02:
> a big up to you, @cainpsycode! After some back and forth it worked for me!
> 
> If someone else wants to build it from source:
> I did that on my Raspberry directly. For me, there was to need to run the following command
> 
> ```
> $ sudo update-alternatives --install /usr/bin/node-gyp node-gyp /opt/node-v10.15.3-linux-armv7l/bin/node-gyp 1
> ```
> 
> Instead I need to run the following commands as sudo in the node-dht-sensor directory
> 
> ```
> $ sudo node-gyp configure
> $ sudo node-gyp build
> ```
> 
> I was also missing the package `node-addon-api` but a simple `npm install node-addon-api` fixed it.
> 
> Once the build process has been completed I could import the self-made build via - in my case with promises -
> 
> ```
> var sensor = require("./lib").promises;
> 
> ... do fancy things here
> ```
> 
> // edit: no I hope that the library does not suffer from the negativ values bug as the Python lib I used to use does ;)

There is no hope. The negative values are wrong.

KillerJulian, 2021-12-20:
I have the same timeout problem when I run the DHT11 sensor on 3.3V. I have changed my setup and use 5V on the sensor. But even there it happens sometimes!

I hope this problem will be solved soon in the LIB

Ricky1966, 2022-04-13:
Hi, sorry for my english, but isn't my first languege.
I've read this discussion, I've try to change latency as @cainpsycode suggest, I've compiled and try, the issue there is still.
Also I, with Python, have the rigth values, so isn't a probe problem.
I've changed voltage to 3v3. Yet same problem.
Someone have fixed this problem?
Ther is a fork of project?
Tks

momenso, 2026-04-18:
The `node-dht-sensor` library does not enable internal pull-ups because the DHT protocol requires sharp electrical edges. The recommended approach is to wire a physical 4.7kΩ - 10kΩ pull-up resistor between the data line and 3.3V for this library to read the pulses accurately. Please note that most DHT sensor modules already provide the pull-up resistor nowadays.

I'm closing this issue due to inactivity, but if the error persists, please reopen the issue and include specific details on how the sensor is wired, along with the versions of the OS, `node-dht-sensor`, Node.js, and your Raspberry Pi model.

closed 2026-04-18 by momenso