momenso/node-dht-sensor #137: Build error on Windows 11

khui0, 2023-11-11:
Attempting to install `npm i node-dht-sensor` on Windows results in a build error. It works fine on my Raspberry Pi, same Node version.

Windows 11 Pro 23H2 Build 22631.2506

```
npm ERR! code 1
npm ERR! path C:\Users\kenny\Desktop\dht11web\node_modules\node-dht-sensor
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c node-gyp build
npm ERR! nothing.c
npm ERR!   win_delay_load_hook.cc
npm ERR!   nothing.vcxproj -> C:\Users\kenny\Desktop\dht11web\node_modules\node-dht-sensor\build\Release\\nothing.lib
npm ERR!   bcm2835.c
npm ERR! C:\Users\kenny\Desktop\dht11web\node_modules\node-dht-sensor\src\bcm2835\bcm2835.c(16,10): fatal  error C1083: Cannot open include file: 'sys/mman.h': No such file or directory [C:\Users\kenny\Desktop\dht11web\node_modules\node-dht-sensor\build\node_dht_sensor.vcxproj]
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.4.0
npm ERR! gyp info using node@20.9.0 | win32 | x64
npm ERR! gyp info spawn C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\MSBuild\Current\Bin\MSBuild.exe
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   'build/binding.sln',
npm ERR! gyp info spawn args   '/clp:Verbosity=minimal',
npm ERR! gyp info spawn args   '/nologo',
npm ERR! gyp info spawn args   '/p:Configuration=Release;Platform=x64'
npm ERR! gyp info spawn args ]
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\MSBuild\Current\Bin\MSBuild.exe` failed with exit code: 1
npm ERR! gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:203:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:514:28)
npm ERR! gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:294:12)
npm ERR! gyp ERR! System Windows_NT 10.0.22631
npm ERR! gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build"
npm ERR! gyp ERR! cwd C:\Users\kenny\Desktop\dht11web\node_modules\node-dht-sensor
npm ERR! gyp ERR! node -v v20.9.0
npm ERR! gyp ERR! node-gyp -v v9.4.0
npm ERR! gyp ERR! not ok
```

khui0, 2023-11-12:
I don't think this package can run on Windows? Is there a way to test it on Windows?

KillerJulian, 2023-11-26:
+1

`Cannot open include file: 'sys/mman.h': No such file or directory`