momenso/node-dht-sensor #138: error install

andy0983, 2023-12-15:
root@DietPi:/home/dietpi# sudo npm install --unsafe-perm -g node-dht-sensor
npm ERR! code 1
npm ERR! path /usr/local/lib/node_modules/node-dht-sensor
npm ERR! command failed
npm ERR! command sh -c node-gyp build
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@10.0.1
npm ERR! gyp info using node@21.4.0 | linux | arm64
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: not found: make
npm ERR! gyp ERR! stack at getNotFoundError (/usr/local/lib/node_modules/npm/node_modules/which/lib/index.js:16:17)
npm ERR! gyp ERR! stack at which (/usr/local/lib/node_modules/npm/node_modules/which/lib/index.js:77:9)
npm ERR! gyp ERR! stack at async doWhich (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:112:22)
npm ERR! gyp ERR! stack at async loadConfigGypi (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:77:7)
npm ERR! gyp ERR! stack at async build (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:35:3)
npm ERR! gyp ERR! stack at async run (/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js:81:18)
npm ERR! gyp ERR! System Linux 6.1.21-v8+
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build"
npm ERR! gyp ERR! cwd /usr/local/lib/node_modules/node-dht-sensor
npm ERR! gyp ERR! node -v v21.4.0
npm ERR! gyp ERR! node-gyp -v v10.0.1
npm ERR! gyp ERR! not ok

andy0983, 2023-12-15:
help

andy0983, 2023-12-15:
npm ERR! code 1
npm ERR! path /usr/local/lib/node_modules/node-dht-sensor
npm ERR! command failed
npm ERR! command sh -c node-gyp build
npm ERR! make: Entering directory '/usr/local/lib/node_modules/node-dht-sensor/build'
npm ERR!   CC(target) Release/obj.target/nothing/node_modules/node-addon-api/src/nothing.o
npm ERR! rm -f Release/obj.target/node_modules/node-addon-api/src/nothing.a Release/obj.target/node_modules/node-addon-api/src/nothing.a.ar-file-list; mkdir -p `dirname Release/obj.target/node_modules/node-addon-api/src/nothing.a`
npm ERR! ar crs Release/obj.target/node_modules/node-addon-api/src/nothing.a @Release/obj.target/node_modules/node-addon-api/src/nothing.a.ar-file-list
npm ERR!   COPY Release/nothing.a
npm ERR!   CC(target) Release/obj.target/node_dht_sensor/src/bcm2835/bcm2835.o
npm ERR!   CXX(target) Release/obj.target/node_dht_sensor/src/node-dht-sensor.o
npm ERR! make: Leaving directory '/usr/local/lib/node_modules/node-dht-sensor/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@10.0.1
npm ERR! gyp info using node@21.4.0 | linux | arm64
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../src/bcm2835/bcm2835.c: In function ‘bcm2835_init’:
npm ERR! ../src/bcm2835/bcm2835.c:1786:44: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
npm ERR!  1786 |                 bcm2835_peripherals_base = (uint32_t *)base_address;
npm ERR!       |                                            ^
npm ERR! ../src/bcm2835/bcm2835.c:1818:84: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
npm ERR!  1818 |       bcm2835_peripherals = mapmem("gpio", bcm2835_peripherals_size, memfd, (off_t)(uint32_t)bcm2835_peripherals_base);
npm ERR!       |                                                                                    ^
npm ERR! ../src/bcm2835/bcm2835.c:1851:84: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
npm ERR!  1851 |       bcm2835_peripherals = mapmem("gpio", bcm2835_peripherals_size, memfd, (off_t)(uint32_t)bcm2835_peripherals_base);
npm ERR!       |                                                                                    ^
npm ERR! make: g++: No such file or directory
npm ERR! make: *** [node_dht_sensor.target.mk:117: Release/obj.target/node_dht_sensor/src/node-dht-sensor.o] Error 127
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:209:23)
npm ERR! gyp ERR! System Linux 6.1.21-v8+
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build"
npm ERR! gyp ERR! cwd /usr/local/lib/node_modules/node-dht-sensor
npm ERR! gyp ERR! node -v v21.4.0
npm ERR! gyp ERR! node-gyp -v v10.0.1
npm ERR! gyp ERR! not ok

ADKMechETECH, 2025-12-28:
Bump to this. I have the same issue on RPI5.  Followed instructions and installed libgpiod-dev after installing Node-Red on a fresh Raspbian image.

momenso, 2026-04-18:
Closing this issue due to inactivity.

If the error persists, please reopen the issue and include the following:
* The complete version of Node.js and npm you are using.
* The specific operating system and hardware (e.g., Raspberry Pi model).
* Detailed steps to reproduce the error.

Note: Based on the logs provided, it appears the `make` tool is missing from your system (`Error: not found: make`). On systems like DietPi or Raspberry Pi OS, you typically need to install build essentials first:
```bash
sudo apt-get update
sudo apt-get install build-essential
```